opentrack/opentrack #725: OpenTrack writes head movement(Yaw, pitch and roll) incorrectly in the csv file

ddrmorais, 2018-01-16:
Hi Guys,

	I am using 360 videos and I need to employ head movement to this kind of videos. So, I use OpenTrack 2.3.9 and FreePie IMU to create an interaction between the Virtual Glass and the PC. The virtual glass transmits the head movements (yaw, pitch and roll)  using the FreePie to the Computer. Thus, the OpenTrack 2.3.9 (Desktop Software) receives this information, prints it in the screen and records it in a csv file. However, the data (yaw, pitch and roll)that is showed in the display is different
from recorded data. Whereas, they should be the same. As you can see in the picture just for a simple example, the yaw is turned in 45 degrees and the opentrack measures it correctly. But when opentrack software writes this infomation in the csv file, the head movements are wrong.
      Does anybody have any idea how to solve this problem?

[image]

sthalik, 2018-01-17:
1. What is your opentrack version?
2. opentrack records the pose during various transformation. You'd want the first six values, they're exactly what the device sends.

sthalik, 2018-01-17:
opentrack records the pose during various transformations. There are various labels here:

- `raw`: what freepie sent, "raw data" window
- `mapped`: what you see in the "game data" window.

sthalik, 2018-01-18:
@ddrmorais how large are the discrepancies?

ddrmorais, 2018-01-18:
The data does not match at all.

I attached 2 files : a Gif with the example of my test and its csv file .

This first simple example intends to verify if Game data (in the screen) and Mapped data(in the csv file) are matching.
[OpenTrackTest.zip](https://github.com/opentrack/opentrack/files/1642781/OpenTrackTest.zip)

sthalik, 2018-01-18:
Here's what LibreOffice draws:

[image]

I see you centered after several seconds. From this point it's offsetted on the Y axis.

It's strange how the maximum yaw value is (-135) rather than close to (-180). Is this the issue?

ddrmorais, 2018-01-19:
I have just found out the error.
I normally open the csv file in Excel, and I dont know what happens that It messes up the numbers. When I opened in the gedit editor I have had no problem with the data (mappedYaw, mappedPitch and ,mappedRoll).

sthalik, 2018-01-19:
Delimiters?

sthalik, 2018-01-19:
Looks like we're god.

closed 2018-01-19 by sthalik