opentrack/opentrack #726: linux compilation error

ljgdasfhk, 2018-01-17:
Scanning dependencies of target opentrack-proto-libevdev
[ 77%] Building CXX object proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/ftnoir_protocol_libevdev.cpp.o
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp: In member function 'virtual module_status evdev::initialize()':
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:111:22: error: 'tr' was not declared in this scope
         return error(tr("Can't open /dev/uinput: %1").arg(buf));
                      ^~
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:111:22: note: suggested alternative: 'tm'
         return error(tr("Can't open /dev/uinput: %1").arg(buf));
                      ^~
                      tm
make[2]: *** [proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/build.make:76: proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/ftnoir_protocol_libevdev.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2769: proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

ljgdasfhk, 2018-01-17:
may be "QWidget::tr(...)" ?
then Ok.

sthalik, 2018-01-17:
I also enabled proto/libevdev always if found.

sthalik, 2018-01-17:
Are there any other Linux build errors?

ljgdasfhk, 2018-01-17:
if i change "tr" to "QWidget::tr(...)" in ftnoir_protocol_libevdev.cpp:111 then no errors.

ljgdasfhk, 2018-01-17:
no errors now

ljgdasfhk, 2018-01-17:
the main window is not displayed

sthalik, 2018-01-18:
It shows for me. Build with `-g` and send me a backtrace
```
thread apply all bt full
```

ljgdasfhk, 2018-01-18:
how can i do it ?

sthalik, 2018-01-18:
Which part?

ljgdasfhk, 2018-01-18:
git clone https://github.com/opentrack/opentrack.git
cd ./opentrack
env LC_ALL=C LANG= LANGUAGE= cmake --build .
...
?

sthalik, 2018-01-18:
`ccmake .`.

Press `t`, add `-g` at the end as an argument to `CMAKE_CXX_FLAGS`. Press `c` then `g` then `ccmake` will close itself. Rebuild and install again.

For debugging a process using `gdb(1)` and obtaining there are too many short tutorials. Just that after it finished initialization after a few seconds, execute

```
thread apply all bt full
```

From the `gdb(1)` prompt and pastebin the contents.

sthalik, 2018-01-18:
Actually for the `ccmake` step there's an easier alternative.

Just run
````
cmake -DCMAKE_CXX_FLAGS='-O2 -g' .
````
from the build directory.

For `gdb(1)` do

```
gdb bin/opentrack # from the install prefix after installation, the file will exist
r
thread apply all bt
```

ljgdasfhk, 2018-01-18:
~]$ gdb /programms/opentrack/bin/opentrack
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /programms/opentrack/bin/opentrack...done.
(gdb) thread apply all bt full
(gdb) thread apply all bt full
(gdb) thread apply

sthalik, 2018-01-18:
Use `r` as a command to start the executable.

`^C` is a shorthand for pressing Control+C.

ljgdasfhk, 2018-01-19:
Reading symbols from ./opentrack...done.
(gdb) r
Starting program: /programms/opentrack/bin/opentrack 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffea666700 (LWP 3570)]
[New Thread 0x7fffcd387700 (LWP 3571)]
[New Thread 0x7fffcb7d9700 (LWP 3572)]
qxt-mini: adding event filter
qxt-mini: fixed x11 autorepeat
^C
Thread 1 "opentrack" received signal SIGINT, Interrupt.
0x00007ffff491791b in poll () from /usr/lib/libc.so.6
(gdb) thread apply all bt full

Thread 4 (Thread 0x7fffcb7d9700 (LWP 3572)):
#0  0x00007ffff491791b in poll () at /usr/lib/libc.so.6
#1  0x00007ffff222b023 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff222b13e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff57c5b71 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/libQt5Core.so.5
#4  0x00007ffff5767d0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff557778e in QThread::exec() () at /usr/lib/libQt5Core.so.5
#6  0x00007fffed5fa416 in  () at /usr/lib/libQt5DBus.so.5
#7  0x00007ffff557cb2d in  () at /usr/lib/libQt5Core.so.5
#8  0x00007ffff3fdc08c in start_thread () at /usr/lib/libpthread.so.0
#9  0x00007ffff4921e1f in clone () at /usr/lib/libc.so.6

Thread 3 (Thread 0x7fffcd387700 (LWP 3571)):
#0  0x00007ffff491791b in poll () at /usr/lib/libc.so.6
#1  0x00007fffd4997d31 in  () at /usr/lib/libusb-1.0.so.0
#2  0x00007ffff3fdc08c in start_thread () at /usr/lib/libpthread.so.0
#3  0x00007ffff4921e1f in clone () at /usr/lib/libc.so.6

Thread 2 (Thread 0x7fffea666700 (LWP 3570)):
#0  0x00007ffff491791b in poll () at /usr/lib/libc.so.6
#1  0x00007fffeeb248e0 in  () at /usr/lib/libxcb.so.1
#2  0x00007fffeeb26679 in xcb_wait_for_event () at /usr/lib/libxcb.so.1
#3  0x00007fffedaf27aa in  () at /usr/lib/libQt5XcbQpa.so.5
#4  0x00007ffff557cb2d in  () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff3fdc08c in start_thread () at /usr/lib/libpthread.so.0
#6  0x00007ffff4921e1f in clone () at /usr/lib/libc.so.6

Thread 1 (Thread 0x7ffff7fa5600 (LWP 3566)):
#0  0x00007ffff491791b in poll () at /usr/lib/libc.so.6
#1  0x00007ffff222b023 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff222b13e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff57c5b71 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/libQt5Core.so.5
#4  0x00007fffedb7a2f2 in  () at /usr/lib/libQt5XcbQpa.so.5
#5  0x00007ffff5767d0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
---Type <return> to continue, or q <return> to quit---
#6  0x00007ffff5770ff8 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#7  0x00007ffff7b9a7d4 in run_window(QApplication&, std::unique_ptr<QWidget, std::default_delete<QWidget> >) (app=..., main_window=...) at /home/sobolev/opentrack/gui/init.cpp:159
        app = <optimized out>
        main_window = <optimized out>
#8  0x00007ffff7b9aad9 in otr_main(int, char**, std::function<QWidget* ()>) (argc=<optimized out>, argv=<optimized out>, make_main_window=...) at /home/sobolev/opentrack/gui/init.cpp:205
        app = <incomplete type>
        t = <incomplete type>
        ret = <optimized out>
#9  0x000055555555f56b in run_application<main(int, char**)::<lambda()> > (fun=..., argv=<optimized out>, argc=<optimized out>) at /home/sobolev/opentrack/gui/init.hpp:15
#10 0x000055555555f56b in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /home/sobolev/opentrack/variant/default/main.cpp:10

ljgdasfhk, 2018-01-19:
compilation does not work again

[ 75%] Building CXX object proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/ftnoir_protocol_libevdev.cpp.o
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp: In member function 'virtual void evdev::pose(const double*)':
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:96:26: error: 'clamp' was not declared in this scope
         int normalized = clamp(value, min_input, max_input);
                          ^~~~~
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:96:26: note: suggested alternative:
In file included from /usr/include/c++/7.2.1/algorithm:62:0,
                 from /usr/include/qt/QtCore/qglobal.h:110,
                 from /usr/include/qt/QtCore/qatomic.h:41,
                 from /usr/include/qt/QtCore/qvariant.h:43,
                 from /usr/include/qt/QtCore/QVariant:1,
                 from /home/sobolev/opentrack-build/proto-libevdev/ui_ftnoir_libevdev_controls.h:12,
                 from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.h:8,
                 from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:1:
/usr/include/c++/7.2.1/bits/stl_algo.h:3737:5: note:   'std::clamp'
     clamp(const _Tp& __val, const _Tp& __lo, const _Tp& __hi, _Compare __comp)
     ^~~~~
make[2]: *** [proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/build.make:76: proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/ftnoir_protocol_libevdev.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2769: proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

ljgdasfhk, 2018-01-19:
$ gdb /programms/opentrack/bin/opentrack
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /programms/opentrack/bin/opentrack...done.
(gdb) r
Starting program: /programms/opentrack/bin/opentrack 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffea664700 (LWP 17234)]
[New Thread 0x7fffcd389700 (LWP 17235)]
[New Thread 0x7fffcb7db700 (LWP 17236)]
qxt-mini: adding event filter
qxt-mini: fixed x11 autorepeat
^C
Thread 1 "opentrack" received signal SIGINT, Interrupt.
0x00007ffff491591b in poll () from /usr/lib/libc.so.6
(gdb) thread apply all bt full

Thread 4 (Thread 0x7fffcb7db700 (LWP 17236)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007ffff2229023 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff222913e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff57c3b71 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/libQt5Core.so.5
#4  0x00007ffff5765d0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff557578e in QThread::exec() () at /usr/lib/libQt5Core.so.5
#6  0x00007fffed5f8416 in  () at /usr/lib/libQt5DBus.so.5
#7  0x00007ffff557ab2d in  () at /usr/lib/libQt5Core.so.5
#8  0x00007ffff3fda08c in start_thread () at /usr/lib/libpthread.so.0
#9  0x00007ffff491fe1f in clone () at /usr/lib/libc.so.6

Thread 3 (Thread 0x7fffcd389700 (LWP 17235)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007fffd4999d31 in  () at /usr/lib/libusb-1.0.so.0
#2  0x00007ffff3fda08c in start_thread () at /usr/lib/libpthread.so.0
#3  0x00007ffff491fe1f in clone () at /usr/lib/libc.so.6

Thread 2 (Thread 0x7fffea664700 (LWP 17234)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007fffeeb228e0 in  () at /usr/lib/libxcb.so.1
#2  0x00007fffeeb24679 in xcb_wait_for_event () at /usr/lib/libxcb.so.1
#3  0x00007fffedaf07aa in  () at /usr/lib/libQt5XcbQpa.so.5
#4  0x00007ffff557ab2d in  () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff3fda08c in start_thread () at /usr/lib/libpthread.so.0
#6  0x00007ffff491fe1f in clone () at /usr/lib/libc.so.6

Thread 1 (Thread 0x7ffff7fa5600 (LWP 17230)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007ffff2229023 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff222913e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff57c3b71 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/libQt5Core.so.5
#4  0x00007fffedb782f2 in  () at /usr/lib/libQt5XcbQpa.so.5
#5  0x00007ffff5765d0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
---Type <return> to continue, or q <return> to quit---
#6  0x00007ffff576eff8 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#7  0x00007ffff7b999a4 in run_window(QApplication&, std::unique_ptr<QWidget, std::default_delete<QWidget> >) (app=..., main_window=...) at /home/sobolev/opentrack/gui/init.cpp:159
        app = <optimized out>
        main_window = <optimized out>
#8  0x00007ffff7b99ca9 in otr_main(int, char**, std::function<QWidget* ()>) (argc=<optimized out>, argv=<optimized out>, make_main_window=...) at /home/sobolev/opentrack/gui/init.cpp:205
        app = <incomplete type>
        t = <incomplete type>
        ret = <optimized out>
#9  0x000055555555f72b in run_application<main(int, char**)::<lambda()> > (fun=..., argv=<optimized out>, argc=<optimized out>) at /home/sobolev/opentrack/gui/init.hpp:15
#10 0x000055555555f72b in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /home/sobolev/opentrack/variant/default/main.cpp:10
(gdb)

sthalik, 2018-01-19:
Note, here's a longer writeup that's probably overkill.

https://gist.github.com/sthalik/47c99ca2367af0c49a337520ac5809cf

sthalik, 2018-01-19:
It's interesting what shows when you press `^C` and do `bt` in case the last backtrace doesn't point to the main window showing.

sthalik, 2018-01-19:
I'm sending you a modified `variant/default/main-window.cpp`. Just replace the file.

https://pastebin.com/mDBeFbQ

sthalik, 2018-01-19:
Replace `variant/default/main-window.cpp` with this paste.

https://pastebin.com/Rs7rC5bu

ljgdasfhk, 2018-01-19:
Starting program: /programms/opentrack/bin/opentrack 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffea664700 (LWP 19607)]
[New Thread 0x7fffcd387700 (LWP 19608)]
[New Thread 0x7fffcb7d9700 (LWP 19609)]
qxt-mini: adding event filter
starting in tray: true
qxt-mini: fixed x11 autorepeat
^C
Thread 1 "opentrack" received signal SIGINT, Interrupt.
0x00007ffff491591b in poll () from /usr/lib/libc.so.6
(gdb) thread apply all bt full

Thread 4 (Thread 0x7fffcb7d9700 (LWP 19609)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007ffff2229023 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff222913e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff57c3b71 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/libQt5Core.so.5
#4  0x00007ffff5765d0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff557578e in QThread::exec() () at /usr/lib/libQt5Core.so.5
#6  0x00007fffed5f8416 in  () at /usr/lib/libQt5DBus.so.5
#7  0x00007ffff557ab2d in  () at /usr/lib/libQt5Core.so.5
#8  0x00007ffff3fda08c in start_thread () at /usr/lib/libpthread.so.0
#9  0x00007ffff491fe1f in clone () at /usr/lib/libc.so.6

Thread 3 (Thread 0x7fffcd387700 (LWP 19608)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007fffd4997d31 in  () at /usr/lib/libusb-1.0.so.0
#2  0x00007ffff3fda08c in start_thread () at /usr/lib/libpthread.so.0
#3  0x00007ffff491fe1f in clone () at /usr/lib/libc.so.6

Thread 2 (Thread 0x7fffea664700 (LWP 19607)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007fffeeb228e0 in  () at /usr/lib/libxcb.so.1
#2  0x00007fffeeb24679 in xcb_wait_for_event () at /usr/lib/libxcb.so.1
#3  0x00007fffedaf07aa in  () at /usr/lib/libQt5XcbQpa.so.5
#4  0x00007ffff557ab2d in  () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff3fda08c in start_thread () at /usr/lib/libpthread.so.0
#6  0x00007ffff491fe1f in clone () at /usr/lib/libc.so.6

Thread 1 (Thread 0x7ffff7fa5600 (LWP 19603)):
#0  0x00007ffff491591b in poll () at /usr/lib/libc.so.6
#1  0x00007ffff2229023 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff222913e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff57c3b71 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /usr/lib/libQt5Core.so.5
#4  0x00007fffedb782f2 in  () at /usr/lib/libQt5XcbQpa.so.5
#5  0x00007ffff5765d0b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
---Type <return> to continue, or q <return> to quit---
#6  0x00007ffff576eff8 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#7  0x00007ffff7b999a4 in run_window(QApplication&, std::unique_ptr<QWidget, std::default_delete<QWidget> >) (app=..., main_window=...) at /home/sobolev/opentrack/gui/init.cpp:159
        app = <optimized out>
        main_window = <optimized out>
#8  0x00007ffff7b99ca9 in otr_main(int, char**, std::function<QWidget* ()>) (argc=<optimized out>, argv=<optimized out>, make_main_window=...) at /home/sobolev/opentrack/gui/init.cpp:205
        app = <incomplete type>
        t = <incomplete type>
        ret = <optimized out>
#9  0x000055555555f7fb in run_application<main(int, char**)::<lambda()> > (fun=..., argv=<optimized out>, argc=<optimized out>) at /home/sobolev/opentrack/gui/init.hpp:15
#10 0x000055555555f7fb in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /home/sobolev/opentrack/variant/default/main.cpp:10

ljgdasfhk, 2018-01-19:
in the settings I turned off the system tray and the main window began to appear at startup.

sthalik, 2018-01-19:
Do you have system tray usage enabled in opentrack's profile?

What environment/window manager are you using?

ljgdasfhk, 2018-01-19:
many identical messages in the console:
void DBusMenuExporterPrivate::fillLayoutItem(DBusMenuLayoutItem*, QMenu*, int, int, const QStringList&): No id for action

ljgdasfhk, 2018-01-19:
I disabled it now.
arch-linux, lxqt

sthalik, 2018-01-19:
Does it come with a working tray?

ljgdasfhk, 2018-01-19:
[image]

sthalik, 2018-01-19:
Does opentrack show in the tray if you enable it?

ljgdasfhk, 2018-01-19:
two or three months ago everything worked

ljgdasfhk, 2018-01-19:
yes!

sthalik, 2018-01-19:
Hold on... So it was in tray all along?

ljgdasfhk, 2018-01-19:
NO! two or three months ago everything worked.

if I run the opentrek and then turn on the tray, then the icon in tray is displayed

ljgdasfhk, 2018-01-19:
qt 5.10

sthalik, 2018-01-19:
Try now...

ljgdasfhk, 2018-01-19:
THX!!
The tray now works as before. everything is fine.

sthalik, 2018-01-19:
Thanks for spotting the bug too ❤️ Please drop a message if something breaks again.

Note, you can comment on closed issues. I get email notifications for all issues and will reopen.

ljgdasfhk, 2018-02-03:
[  3%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/shm.cpp.o
/home/sobolev/opentrack/compat/shm.cpp:161:6: error: 'secattr' has not been declared
 void secattr::cleanup()
      ^~~~~~~
/home/sobolev/opentrack/compat/shm.cpp: In function 'void cleanup()':
/home/sobolev/opentrack/compat/shm.cpp:163:9: error: 'pEveryoneSID' was not declared in this scope
     if (pEveryoneSID)
         ^~~~~~~~~~~~
/home/sobolev/opentrack/compat/shm.cpp:164:9: error: 'FreeSid' was not declared in this scope
         FreeSid(pEveryoneSID);
         ^~~~~~~
/home/sobolev/opentrack/compat/shm.cpp:165:9: error: 'pACL' was not declared in this scope
     if (pACL)
         ^~~~
/home/sobolev/opentrack/compat/shm.cpp:166:9: error: 'LocalFree' was not declared in this scope
         LocalFree(pACL);
         ^~~~~~~~~
/home/sobolev/opentrack/compat/shm.cpp:166:9: note: suggested alternative: 'QLocale'
         LocalFree(pACL);
         ^~~~~~~~~
         QLocale
/home/sobolev/opentrack/compat/shm.cpp:167:9: error: 'pSD' was not declared in this scope
     if (pSD)
         ^~~
/home/sobolev/opentrack/compat/shm.cpp:168:9: error: 'LocalFree' was not declared in this scope
         LocalFree(pSD);
         ^~~~~~~~~
/home/sobolev/opentrack/compat/shm.cpp:168:9: note: suggested alternative: 'QLocale'
         LocalFree(pSD);
         ^~~~~~~~~
         QLocale
/home/sobolev/opentrack/compat/shm.cpp:169:5: error: 'success' was not declared in this scope
     success = false;
     ^~~~~~~
/home/sobolev/opentrack/compat/shm.cpp:169:5: note: suggested alternative: 'eaccess'
     success = false;
     ^~~~~~~
     eaccess
/home/sobolev/opentrack/compat/shm.cpp:170:5: error: 'pSD' was not declared in this scope
     pSD = nullptr;
     ^~~
/home/sobolev/opentrack/compat/shm.cpp:171:5: error: 'pEveryoneSID' was not declared in this scope
     pEveryoneSID = nullptr;
     ^~~~~~~~~~~~
/home/sobolev/opentrack/compat/shm.cpp:172:5: error: 'pACL' was not declared in this scope
     pACL = nullptr;
     ^~~~
/home/sobolev/opentrack/compat/shm.cpp: At global scope:
/home/sobolev/opentrack/compat/shm.cpp:175:1: error: 'secattr' does not name a type; did you mean 'qecvt_r'?
 secattr::secattr(DWORD perms) : success(true), pSD(nullptr), pEveryoneSID(nullptr), pACL(nullptr)
 ^~~~~~~
 qecvt_r
/home/sobolev/opentrack/compat/shm.cpp:236:1: error: 'secattr' does not name a type; did you mean 'qecvt_r'?
 secattr::~secattr()
 ^~~~~~~
 qecvt_r
make[2]: *** [compat/CMakeFiles/opentrack-compat.dir/build.make:295: compat/CMakeFiles/opentrack-compat.dir/shm.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:609: compat/CMakeFiles/opentrack-compat.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

sthalik, 2018-02-05:
@ljgdasfhk I fixed this and a warning. Thank you for the reports.

Can you change your name to something else (a nickname or real name) so I can credit you in the changelog? These look like randomly typed keys.

sthalik, 2018-02-05:
Or just give me a name I can credit you under.

ljgdasfhk, 2018-02-05:
Artem Sobolev

ljgdasfhk, 2018-02-05:
[ 75%] Generating opentrack-wrapper-wine.exe.so
In file included from /home/sobolev/opentrack/compat/shm.h:22:0,
                 from /home/sobolev/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx:15:
/home/sobolev/opentrack/compat/macros.hpp:3:10: fatal error: QCoreApplication: No such file or directory
 #include <QCoreApplication>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
winegcc: g++ failed
make[2]: *** [proto-wine/CMakeFiles/wine-wrapper.dir/build.make:63: proto-wine/opentrack-wrapper-wine.exe.so] Error 2
make[1]: *** [CMakeFiles/Makefile2:3208: proto-wine/CMakeFiles/wine-wrapper.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

sthalik, 2018-02-05:
Is it fixed in the newest build? You need to run `cmake .` first in the build directory.

ljgdasfhk, 2018-02-05:
I do not know.
i run cmake . first in the build directory.
new error:

[ 35%] Building CXX object logic/CMakeFiles/opentrack-logic.dir/pipeline.cpp.o
/home/sobolev/opentrack/logic/pipeline.cpp:15:10: fatal error: compat/nan.hpp: No such file or directory
 #include "compat/nan.hpp"
          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [logic/CMakeFiles/opentrack-logic.dir/build.make:179: logic/CMakeFiles/opentrack-logic.dir/pipeline.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1916: logic/CMakeFiles/opentrack-logic.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

ljgdasfhk, 2018-02-05:
and about include QCoreApplication

I have two directories qt and qt4. And in both there is QtCore. 
This may be a problem

sthalik, 2018-02-05:
`compat/nan.hpp` is now gone. It was broken on Windows anyway.

ljgdasfhk, 2018-02-05:
[ 36%] Building CXX object logic/CMakeFiles/opentrack-logic.dir/work.cpp.o
/home/sobolev/opentrack/logic/work.cpp: In lambda function:
/home/sobolev/opentrack/logic/work.cpp:68:55: error: 'using element_type = class gui_tracker_impl::pipeline {aka class gui_tracker_impl::pipeline}' has no member named 'set_center'; did you mean 't_center'?
         key_tuple(s.key_center1, [&](bool) { tracker->set_center(); }, true),
                                                       ^~~~~~~~~~
                                                       t_center
/home/sobolev/opentrack/logic/work.cpp: In lambda function:
/home/sobolev/opentrack/logic/work.cpp:69:55: error: 'using element_type = class gui_tracker_impl::pipeline {aka class gui_tracker_impl::pipeline}' has no member named 'set_center'; did you mean 't_center'?
         key_tuple(s.key_center2, [&](bool) { tracker->set_center(); }, true),
                                                       ^~~~~~~~~~
                                                       t_center
/home/sobolev/opentrack/logic/work.cpp: In lambda function:
/home/sobolev/opentrack/logic/work.cpp:74:53: error: 'using element_type = class gui_tracker_impl::pipeline {aka class gui_tracker_impl::pipeline}' has no member named 'toggle_zero'
         key_tuple(s.key_zero1, [&](bool) { tracker->toggle_zero(); }, true),
                                                     ^~~~~~~~~~~
/home/sobolev/opentrack/logic/work.cpp: In lambda function:
/home/sobolev/opentrack/logic/work.cpp:75:53: error: 'using element_type = class gui_tracker_impl::pipeline {aka class gui_tracker_impl::pipeline}' has no member named 'toggle_zero'
         key_tuple(s.key_zero2, [&](bool) { tracker->toggle_zero(); }, true),
                                                     ^~~~~~~~~~~
/home/sobolev/opentrack/logic/work.cpp: In lambda function:
/home/sobolev/opentrack/logic/work.cpp:77:63: error: 'using element_type = class gui_tracker_impl::pipeline {aka class gui_tracker_impl::pipeline}' has no member named 'set_enabled'; did you mean 'set_toggle'?
         key_tuple(s.key_toggle_press1, [&](bool x) { tracker->set_enabled(!x); }, false),
                                                               ^~~~~~~~~~~
                                                               set_toggle
/home/sobolev/opentrack/logic/work.cpp: In lambda function:
/home/sobolev/opentrack/logic/work.cpp:78:63: error: 'using element_type = class gui_tracker_impl::pipeline {aka class gui_tracker_impl::pipeline}' has no member named 'set_enabled'; did you mean 'set_toggle'?
         key_tuple(s.key_toggle_press2, [&](bool x) { tracker->set_enabled(!x); }, false),
                                                               ^~~~~~~~~~~
                                                               set_toggle
make[2]: *** [logic/CMakeFiles/opentrack-logic.dir/build.make:299: logic/CMakeFiles/opentrack-logic.dir/work.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1916: logic/CMakeFiles/opentrack-logic.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

sthalik, 2018-02-05:
It builds for me cleanly on the `unstable` branch now.

ljgdasfhk, 2018-02-05:
THX!!
everything works well

sthalik, 2018-02-08:
Please post when there are issues again.

ljgdasfhk, 2018-02-08:
Today there are no problems.

ljgdasfhk, 2018-03-01:
Scanning dependencies of target opentrack-version
[  0%] Building CXX object CMakeFiles/opentrack-version.dir/version.cpp.o
[  0%] Linking CXX static library libopentrack-version.a
[  0%] Built target opentrack-version
Scanning dependencies of target i18n
make[2]: *** No rule to make target 'tracker-udp/lang/nl_NL.ts', needed by 'nl_NL.qm'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/i18n.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

sthalik, 2018-03-03:
Does it help to:

```
ninja mrproper
cmake .
```

Note, this will NOT delete your cache.

ljgdasfhk, 2018-03-12:
ninja: error: loading 'build.ninja': No such file or directory

sthalik, 2018-03-13:
```
cmake .
ninja mrproper
cmake .
```

ljgdasfhk, 2018-03-13:
git clone https://github.com/opentrack/opentrack.git
cd ./opentrack
cmake .
CMake Error at cmake/opentrack-check-build-directory.cmake:6 (message):
  source directory has to be different than build directory
Call Stack (most recent call first):
  CMakeLists.txt:25 (include)

-- Configuring incomplete, errors occurred!

sthalik, 2018-03-14:
No.

You should have `CMakeCache.txt` somewhere already. Not in the toplevel directory. Go there and execute the commands.

If you don't, use `cmake-gui` or `ccmake` to configure the build directory anew.

ljgdasfhk, 2018-03-14:
cd ./opentrack-build
cmake .
ninja mrproper
ninja: error: loading 'build.ninja': No such file or directory

sthalik, 2018-03-14:
Nothing changed. You don't have to use ninja.

Either use `cmake /path/to/opentrack/source/dir` or use `cmake-gui`.

ljgdasfhk, 2018-03-14:
ok
git clone https://github.com/opentrack/opentrack.git
cmake .
make
Scanning dependencies of target opentrack-version
[ 0%] Building CXX object CMakeFiles/opentrack-version.dir/version.cpp.o
[ 0%] Linking CXX static library libopentrack-version.a
[ 0%] Built target opentrack-version
Scanning dependencies of target i18n
make[2]: *** No rule to make target 'tracker-udp/lang/nl_NL.ts', needed by 'nl_NL.qm'. Stop.
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/i18n.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

[120 more replies]