opentrack/opentrack #728: csv replay?

b1he1, 2018-01-21:
Is there a feature where opentrack can replay previously recorded csv data? E.g. In the input drop-down box, select csv data and select the file, clicking start would take the csv data into the output. Thanks.

sthalik, 2018-01-21:
Not internally. Use Python and UDP input.

sthalik, 2018-01-21:
Note, .csv is for debugging by developers. That's why it has so many columns for each DOF. It's not a user feature.

closed 2018-01-21 by sthalik

b1he1, 2018-01-21:
Okay, so I'm wondering if there is an 'action replay' feature that doesn't have to be based off of csv files but still be able to save a file for later usage. 

I had a look on the Python example from issue 13&547 and was able to use that to print the csv file onto the Python shell but I'm not sure how I would get the UDP part to work. Unfortunately I can't really code 🙁

sthalik, 2018-01-21:
You'd want to use the `struct` module to send parsed values as a 6-element array of type `double` for each element.