opentrack/opentrack #732: Problem about identifying in OpenTrack.

ddrmorais, 2018-01-22:
Hi Guys,

 
     I am facing a problem about identifying angles in OpenTrack.
 I proposed the task of turning 360 degrees clockwise and after that 360 degrees anti clockwise. The result of the this test is attached.

So, when I started turning from 0 to 90 degrees I had no problem. From 90 to 180, the yaw data stuck to -180 degrees which is not the expected result. The same issue happened in the anti clockwise. 

I believe it is just a setting problem.

Any idea how I could solve this problem?

   Regards,
[turningProblem.zip](https://github.com/opentrack/opentrack/files/1651965/turningProblem.zip)

Len62, 2018-01-22:
I may be wrong but I'm pretty sure that the maximum yaw in either direction is limited to 180. Are you trying to get a full 360 by just moving clockwise or counterclockwise? That would be like rotating your head a full 360.

ddrmorais, 2018-01-22:
Hi,

    I am not trying get a full 360 at once but I want to complete 360
degrees.  I also know the maximum yaw is limited to 180. So, I want to
measure  from 0 to 180 and from -180 to 0 clockwise.
    If you could please look at the Gif attached, I have posted the
movements I want to measure. If you look at the video, It might be easier
to understand the issue explained.
    Anyway, from 90 to 180 degrees the opentrack gets lost and It does
measure the head movements correctly (yaw. pitch and roll).

Len62, 2018-01-23:
Okay, am I correct in assuming you are using VR? If so, are you trying to get 180 degrees of movement in OT with moving your head physically 90 degrees? Also, what are you using as a tracking device?

sthalik, 2018-01-23:
Please make a video showing the opentrack window during that regular circular motion. On the same video show the mapping window with how the red dot repositions itself durign the rotation. Switch to the tab that corresponds to the particular rotational degree of freedom.

sthalik, 2018-01-23:
As for opentrack going out of whack, I'm gonna need to run some tests with you but let's see the video first.

ddrmorais, 2018-01-23:
@ Len62, I am using VR glasses and I trying to make sure that Glasses are turning to each angle correctly. Besides Each angle needs to be shown correctly in the screen. And it does not happens between 90 and 180 degrees. Physically, it is not possible to move your head more than 90 degrees if are stopped. But If you use a chair that has a turning device. We can do it.

ddrmorais, 2018-01-23:
@sthalik I will make the video

ddrmorais, 2018-01-23:
@stanislaw,
As you can see in this video, the problem happened between -180 and -125
degrees. Which means that every time I turn on the VR glasses the issue
happens with different angles randomly.
But the fact is, the yaw stuck in 180 degrees even if I moved the classes
between -180 and -125 degrees.

First, I would like to understand if the problem is clear for you.
If it is , what would you advise me to do?
If it is not, please ask me that I will be pleased to answer your questions.
​
 VID_20180123_155530912.zip
<https://drive.google.com/file/d/1qUsnNAzgtzUfz7hBfiZqMmWii5gmzD-Q/view?usp=drive_web>
​

sthalik, 2018-01-24:
@ddrmorais 

I've had other users come with this problem before.

The solution may be that the tracker should do centering. Please specify the name of your "input" selection, i.e. the tracker.