opentrack/opentrack #733: How can i send my data to OpenTrack via UDP

umutbaser, 2018-01-22:
`import socket
import time
import struct
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

server_address = '0.0.0.0'

server_port = 5555

server = (server_address, server_port)
sock.bind(server)
UDP_IP = "127.0.0.1"
UDP_PORT = 4242
sock2 = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

print("Listening on " + server_address + ":" + str(server_port))

while True:
        localtime = time.asctime( time.localtime(time.time()) )
        message, address = sock.recvfrom(8192)
        
        #try:
 #           print(message.decode("utf-8"))
        a = message.decode("utf-8")
        x,y,z,yaw,pitch,roll = a.split(", ")
        x1=float(x)
        x2=float(y)
        x3=float(z)
        yaw1=float(yaw)
        yaw2=float(pitch)
        yaw3=float(roll)
        msg = "("+x+", "+y+", "+z+(", ")+yaw+", "+pitch+", "+roll+")"
        msgb = struct.pack("dddddd", x1,x2,x3,yaw1,yaw2,yaw3)
        sock2.sendto(msgb, (UDP_IP, UDP_PORT))
        print(struct.unpack("dddddd",msgb))`

My code is working in debug windows but in opentrack it is not changing

sthalik, 2018-01-23:
1. Quote using three tilde key backticks, then code, then three tilde backticks again.
2. You're receiving data rather than sending it.

sthalik, 2018-01-23:
1. Quote the code according to Markdown syntax, i.e. three tilde backticks on a separate line, code, then three backticks on a separate line.
2. It's outside the scope of the project to help with basic programming tasks, but I might help you if you present a concise problem.

cpuwolf, 2018-02-12:
double last_recv_pose[6], last_recv_pose2[6];

closed 2018-08-04 by sthalik