opentrack/opentrack #736: How can i send my HeadMovement data  to my udpClient from opentrack via udp?

fuxingyang, 2018-01-26:
when i set the output at udp over network,library load error！
[image]

fuxingyang, 2018-01-26:
when i set the output at udp over network,library load error！Can you give me some ideas?Thank you

sthalik, 2018-01-26:
The development build has descriptive failure messages. 2.4.0 hasn't yet been released.

I recommend using today's build. Things are quiet right now and it should be as stable as 2.3.9, if not more.

The dev builds are always on <<http://opentrack.misaki.pl>>.

- - -

[image]

closed 2018-08-04 by sthalik