opentrack/opentrack #739: Huge roll data offset

Arthipex, 2018-01-26:
Hi everybody

After having built my own tracker and having successfully operated it for a couple of months with opentrack 2.3, a friend of mine got into flight simming and I agreed to build him his own. 

We're both using the PS3eye with an added IR-bandpass filter and our 3-point clips have the same dimensions (upper and lower width 28mm, upper height 40mm, lower height 57mm).

I feel like I set up everything the right way, however there are some issues with his raw data. Yaw and pitch seem reasonable, however roll appears to be total garbage (around 109°, while camera and led-clip are all oriented correctly). This makes it absolutely impossible to calibrate the model position.

I'm running out of ideas and hope you can help me.

Thanks in advance,
Chris

sthalik, 2018-02-05:
@Arthipex show your model settings and camera frame please. Disable calibration for the time being if you have any.

Arthipex, 2018-02-05:
@sthalik Thank you for your reply. I was able to fix this myself a couple of days ago. It turned out that the camera driver was broken in some way - a proper re-installation solved the issue.

closed 2018-02-08 by sthalik