opentrack/opentrack #740: Opentrack and bitdefender virus alert

M-DF, 2018-01-26:
I have upgraded to Bitdefender 2018 and now on install of the current opentrack it is quarantined as infected with Gen.variant.razy.246207. This has to be a false positive but has anyone else had the problem. I am tempted to just set up an exclusion for the folder but thought I would post here first just in case. 

Mike

sthalik, 2018-01-27:
Yes, see #734.

There's no virus in opentrack's libraries. I know what's in them.

M-DF, 2018-01-27:
Hi. Thanks for the reply. That's what I figured. I contacted bitdefender and have excluded the folder from scan til they fix it. Mike.

sthalik, 2018-02-05:
It's a problem. We're releasing `2.4.0` some time soon. It's not feasible for them to whitelist each build in time, including testing builds.

The real reason is that the `Gen.variant.razy.246207` signature they made is wrong. See this, for a Steam game:

> Bitdefender reports RescueTeam4.exe as virus Gen:Variant.Razy.106858 <<https://steamcommunity.com/app/715090/discussions/0/1489992080501475276/>>

sthalik, 2018-02-28:
@M-DF It should be fine for the `2.3.10` release.

sthalik, 2018-02-28:
Please confirm whether this is "solved" for now on.

M-DF, 2018-02-28:
Hi. I just made exceptions in Bitdefender once I knew it was a false
positive.No further trouble.

Regards

Mike

closed 2018-05-14 by sthalik