opentrack/opentrack #745: Opentrack crashing in connection to Edtracker

Neuse87, 2018-02-08:
Hi, i just found an old problem without a solution here so i´d like to bring this up again.

When i use opentrack with the EDtracker it seems to crash (not responding for minutes) when opening or starting the tracking when the EStracker software is already running.
Tried different versions of both. Always the same. 
Both of them alone run fine. If i start OT and start tracking and then start edsoftware it seems to work till close them. Restarting OT then bugs out again. This seems to slow down opening times of other program aswell as soon it bugs out.

As i said there is already a thread from 2016 that exactly describes the issue.
Just couldnt find a solution there.

sthalik, 2018-02-09:
It looks like a conflict with each program overwriting each other's `.dll` path in the registry. Each of the two programs can't work with the `.dll` from the other one.

You can use the EDTracker device with opentrack directly however.

Neuse87, 2018-02-09:
Thanks for reply. So i just calibrate in ED Tracker Gui and then close it, restart everything and just use Opentrack? Or do i have to calibrate it in Opentrack somehow.

sthalik, 2018-02-09:
No need to calibrate it in opentrack.

Neuse87, 2018-02-13:
after a lot of testint....if i unplug all my oculus rift equipment it all seems to work as intended. How could this all be connected?

sthalik, 2018-02-14:
@Neuse87 does ED use an HMD to the exclusion of all other tracking sources in case it's connected?

closed 2019-02-11 by sthalik