opentrack/opentrack #748: opentrack now has lang RU and NL, can we add Chinese support?

cpuwolf, 2018-02-12:
I would love to support Chinese related portion. I think I could do it from module to module, and tracker/wii is just a beginning.

#746

sthalik, 2018-02-12:
Please see 6bb50c04e5c40a5e036e41dde11b1660c3b077f6. I fixed translation-related issues that we had.

I also added `zh_CN` to the language list in 31d9768413199f388c216d92d9db5158ce8785cf and then running `make install` created the `.ts` files automatically.

Finally, I fixed a bug where `make clean` deleted all translation files.

sthalik, 2018-02-12:
You should be able to do translations from now on. To update based on code changes, run one of the `i18n`, `all` or `install` targets.

It's best to run `QObject::tr` as long as you're within some `QObject` subclass.  See also `_(str, ...)` for when necessary to translate outside `QObject`.

Does it work now?

cpuwolf, 2018-02-12:
until latest commit: tracker/wii: silence #pragma deprecated

there is build error with target "INSTALL" in MSVC

1>------ Build started: Project: i18n, Configuration: Release Win32 ------
1>Building Custom Rule F:/works/GitHub/opentrack/CMakeLists.txt
1>CMake does not need to re-run because F:/works/GitHub/output/opentrack/CMakeFiles/generate.stamp is up-to-date.
1>Running lrelease for nl_NL
1>CUSTOMBUILD : lrelease error : Cannot open F:/works/GitHub/output/opentrack/variant/default/lang/nl_NL.ts: 系统找不到指定的路径。
1>Done building project "i18n.vcxproj" -- FAILED.

sthalik, 2018-02-12:
What happens if you run the `i18n` target manually?

This is strange because `/variant/default/lang/nl_NL.ts` exists in the repository.

sthalik, 2018-02-12:
I'm able to run:

```
dev/opentrack/build % find . -name '*.ts' -delete
dev/opentrack/build % msvc ninja install
```

While there are subtle differences between build systems, it should work after solution reload, `clean` or `mrproper`, then `install`. It should also work on following installs. Of course if you still have an issue it's unacceptable and I need to fix it.

cpuwolf, 2018-02-12:
I created a new folder, and fetched latest commit. now same error with target "build"

1>------ Build started: Project: i18n, Configuration: Release Win32 ------
1>Building Custom Rule F:/works/GitHub/opentrack/CMakeLists.txt
1>CMake does not need to re-run because F:/works/GitHub/output/opentrack/CMakeFiles/generate.stamp is up-to-date.
1>Running lrelease for nl_NL
1>CUSTOMBUILD : lrelease error : Cannot open F:/works/GitHub/output/opentrack/variant/default/lang/nl_NL.ts: 系统找不到指定的路径。
1>Done building project "i18n.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 34 up-to-date, 0 skipped ==========

[image]

I checked the path

F:/works/GitHub/output/opentrack/variant/default/lang/nl_NL.ts

as I show above, there is no folder '''lang'''

sthalik, 2018-02-13:
Does it work now? I was able to build from a `.sln`.

cpuwolf, 2018-02-14:
it works now :+1: 

then what should I expect to see the final result. because what I can see is still English during runtime

sthalik, 2018-02-16:
Works for me now.

closed 2019-02-11 by sthalik