opentrack/opentrack #749: game status after .dll changes

sthalik, 2018-02-16:
#### Background

Starting in 2.4.0, the `freetrack` protocol has an important fix that's was postponed for years. Code changes mean bugs. Please help with testing games you play after the change.

#### Preparation

1. Please perform testing only with builds labelled `20180216` and later. The builds are available at <<http://opentrack.misaki.pl>> as always.
2. Since you just downloaded the build, switch to Freetrack output and start, then stop the tracker. Otherwise the old `.dll` will be used.
3. Always use the dll's from the archive when submitting titles. It's fine to use older dll's for other purposes but for this purpose always use the archive's dll's
4. If running another build, do the "start/stop" thing from step 2 again.

#### The testing procedure

1. Perform all steps in the "preparation" section.
2. Start the game with tracking stopped. You should be able to control the view with mouse, if the game supports it.
3. Start tracking. Head tracking should work.
4. Stop tracking, exit and start the game again. You should have mouse control as in step 2.
5. Start tracking, start the game. Tracking should work.
6. Stop tracking but keep the game open. You should regain mouse control as in step 2.
7. Start tracking again. You should regain head tracking.

These are quite a few steps but it checks a few different things that didn't work before. You don't need to understand the meaning behind the steps to submit games. I can send an explanation if necessary.

#### Reports

Please specify game titles and whether it works according to the testing procedure. If something's broken, specify step number where it fails and/or a comment.

#### Problems with Arma BattlEye

Before testing is complete and the `.dll` submitted, BattlEye won't work with head tracking. You can copy over files `NPClient.dll` and `NPClient64.dll` from the `2.3.9` portable build into the `modules/` subdirectory of the testing build. They have the old bug however.

### Game reports

- American Truck Simulator (@sthalik) -- fails step 7
- Arma 3 (@sthalik) -- all works but will recheck
- Falcon BMS (@sthalik) -- all works but will recheck

Len62, 2018-02-17:
DCS World 2.5 Open Beta -- ok

JoeSchr, 2018-04-09:
he just tried it with the `opentrack-test-20180316_01.7z` which opens up to be `opentrack 2.3.49-plus-48-g7fa77424` with **Arma 3**

I did the preparation steps. Although didn't really understand step 3 & 4, so I just stop/started tracker a few times and located `NPClient.dll` via `freetrack->hammer symbol->locate dll`.

Then I tried it, didn't step 3 didn't work. exited game. also didn't work. basically didn't work at all. The tracker also didn't show up under controller. 

Did work:
* `Version 2.3.9` 
* `Version 2.3.10`

Also didn't work:
* `opentrack-test-20180312.7z`
* `opentrack-test-20180216_11.7z`

sthalik, 2018-04-10:
It only works without BattlEye before it's ready.

JoeSchr, 2018-04-10:
ok, thats really bad then?

sthalik, 2018-04-11:
No, it'll be taken care of.

JoeSchr, 2018-04-11:
Oh, awesome, thank you!

sthalik, 2018-05-14:
I'm thinking of ditching the whole plan. There's more anticheat software that added opentrack's libraries on their own. The bug disabling mouse view seems insignificant compared to the transition period.

closed 2018-05-22 by sthalik