opentrack/opentrack #750: Multiple Version Installs

Len62, 2018-02-17:
Not really an issue per se, but I normally have 2 or 3 versions of opentrack installed at any given time. Does this have any impact with regards to the NPClient.dll issue? Does it have any impact with information stored in the registry? I was looking at the registry entries and one was pointing to an ini file that no longer exists. Would if be safe to delete all of the registry entries and start fresh with the 20180216_11 version thereby forcing new registry entries?

sthalik, 2018-02-17:
Hey @Len62,

> Does this have any impact with regards to the NPClient.dll issue? Does it have any impact with 
> information stored in the registry

Yes, you must start/stop using freetrack protocol prior to starting games for NPClient change to take effect. See registry location `HKEY_CURRENT_USER\Software\NaturalPoint\NATURALPOINT\NPClient Location\Path`.

Upon starting the freetrack protocol, the registry keys are written. They apply to games started from then on.

Don't have games running in the background when doing this start/stop, since the game will use old NPClient and opentrack installation expects its own bundled version.

Please consider whether you can provide data to #749. As the "pipeline rework" project, bugs are taken care of and the reporting user credited in changelogs.

> Would if be safe to delete all of the registry entries and start fresh with the 20180216_11 version 
> thereby forcing new registry entries?

If you use "game detection", this loses its saved data. It's stored globally, since it's profile-agnostic.

I haven't had any need of manually changing registry entries at all. It's more likely a program bug if you encounter any runtime problems.

> I was looking at the registry entries and one was pointing to an ini file that no longer exists. 

opentrack will deal with that, switching to `default.ini`. there's no failure mode due to nonexistent .ini saved as default, or in game detector. The only way you can break it is if the `Documents/opentrack-2.3` directory is made non-readable, non-writable, or nonexistent and prohibited from being created. In this case opentrack will show an error message and exit on startup.

sthalik, 2018-02-17:
Here are my registry settings. You should be able to control them from the user interface. Otherwise it's a bug!

[image]

Len62, 2018-02-17:
Here's a look at reg entries for NPClient and opentrack. I circled the old OT entry I was talking about. In the NPClient entry the path is not being written to the registry. I asked because I still cannot get freetrack to work at all with IL2-BOS or DCSW. 

[image]

[image]

sthalik, 2018-02-17:
It's 99% that you selected "disable npclient, enable freetrack" in the profile. The "path" entry getting set to empty is the symptom.

Old entries are harmless and unused. You can remove the keys and they'll get recreated if you wish.

sthalik, 2018-02-17:
IL-2 BOS works (didn't do full test). The titles absolutely require "NPClient Location" set. Also if you ever set `old_trackir = 1` for BOS, disable it.

Len62, 2018-02-17:
Okay, did some checking and the correct paths are being written to the registry based on the selected output. BOS setting is correct. However, I can still not get the freetrack protocol to work with BOS. Freetrack protocol set, tracker running and working, but no head movement in game.

albertolg, 2018-02-17:
@Len62 just for curiosity, why do you want to use the freetrack protocol if the trackir protocol works fine?

Len62, 2018-02-17:
Just to help with troubleshooting for sthalik.

sthalik, 2018-02-17:
The protocol is called "freetrack" in opentrack but `NPClient` aka "trackir protocol" is what BOS is using. There's no support for "hide trackir, use freetrack" in BOS. Use "enable both".

Look for the game's configuration file and if there's any "old_trackir = 1", remove it. Also see if you get the game name on opentrack's title bar.

Len62, 2018-02-17:
Ok, I was under the impression freetrack should work in BOS. So how do you test: #749 "Since you just downloaded the build, switch to Freetrack output and start, then stop the tracker. Otherwise the old .dll will be used." Maybe I need some more coffee this morning.

albertolg, 2018-02-17:
@sthalik , it's offtopic, but have you changed the behavior of the close button on purpose so it minimize in the traybar? sometimes I forgot and have multiple opentrack opened at the same time :-)

sthalik, 2018-02-17:
@albertolg you're right. That was the old behavior and it made sense. 👍

sthalik, 2018-02-17:
@Len62 the matter is muddled ever since 2012 before the fork, mostly on purpose due to matters related to freetrack team dissolution... I don't plan to surrender or run if push comes to shove.

To be clear, we're not testing `freetrackclient.dll` at all because it didn't change. It's only `npclient.dll` and `npclient64.dll` that are affected by the change.

I changed the testing instructions a bit to be more clear.

Len62, 2018-02-17:
So for testing "enable both" should be selected when conducting the test?

sthalik, 2018-02-17:
Yes.

Len62, 2018-02-17:
Got it! Checked DCS World 2.5 Open Beta and it was ok. Added comment to #749 thread.

closed 2018-02-17 by Len62