opentrack/opentrack #752: experimental bild 20180216_11 crashes when starting before Il-2 Sturmovic Forgotten Battles

Seamas73, 2018-02-23:
When I start opentrack testing build 20180216_11 and then start "Il-2 Sturmovik: Forgotten Battles" game shows error window with nearly this content: " program NPClient.dll not made to work in Windows or broken. Try to reinstall it". And then opentrack crashes. I solved this problem by replacing NPClient.dll with the dll from latest stable release.

sthalik, 2018-02-24:
It's a 64-bit binary. Is it fully patched? Make sure to disable `old_trackir` in IL2 config.

sthalik, 2018-02-24:
To explain -- it should be opening `NPClient64.dll`.

Seamas73, 2018-02-24:
I told not about new "Il2Sturmovik: Battle over X" game, I told about "good old" "Il-2 Sturmovik 1946: Forgotten Battles" game. Old game is 32bit, with all last patches installed. And in it config file don't exist "old_trackir" parameter, there is only "TrackIRUse" parameter.
On stable releases of opentrack old "Il-2" didn't show errors with NPClient.dll and didn't cause crash of opentrack.

sthalik, 2018-02-24:
I'll look into it. For now use today's 2.3.10-pre from the same site.

Seamas73, 2018-02-24:
> For now use today's 2.3.10-pre from the same site.

It works fine

sthalik, 2018-02-25:
With the old build, it can be Windows Defender or SmartScreen. Please check with antivirus disabled.

closed 2018-08-03 by sthalik