opentrack/opentrack #753: PT calibration and clip angle

Simon-Dainty, 2018-02-26:
Is the angle of the mounted clip taken into consideration when calibrating the Point Tracker?  For instance, if the clip is at -25 degs from the vertical does the calibration algorithm take the off-axis into account - and is it even an issue at all?

Simon-Dainty, 2018-02-28:
The IR clip I use is mounted to my headphone's head band (with a velcro strap).  The head band is curved, therefore the IR clip is mounted at a diagonal.  Further more, I'm struggling to calibrate the clip where the roll axis doesn't dramatically introduce yaw and/or pitch.  I've followed the wiki calibration guide and video for what it's worth.

Is the angle of my clip affecting the calibration?

sthalik, 2018-03-03:
> Is the angle of my clip affecting the calibration?

Yes. Currently there's no fix for that. We need a second calibrator for ypr/xyz correlation.

sthalik, 2018-03-03:
I have working code for this but user interface is the problem.

Simon-Dainty, 2018-03-04:
I can appreciate that.  Programmatically, I suppose an imaginary line plotted through the sensors would provide one definite [vertical] axis (that could be used to calculate the angle between some assumed other [horizontal] axis).  An opportunity to manually tweak that would also be beneficial.  Or just a spinner, that the user can employ to define the offset of the angle, again using some on-screen indicator to represent the present angle.

sthalik, 2018-03-07:
I compute correlations. The procedure will look like, ask user to perform given (say, yaw) motion and see how other axis change. From all samples draw the correlation matrix.

sthalik, 2018-03-07:
So here's a piece of the correlation matrix as the code works now. I performed a yaw motion. Removed non-yaw stuff to be less confusing.

Note, it's not connected to tracking stuff.

```
                    x         y         z       yaw     pitch      roll
         x                                   -0.107                    
         y                                    0.026                    
         z                                   -0.215                    
       yaw                                    1.000                    
     pitch                                   -0.070                    
      roll                                   -0.238                    
```

sthalik, 2018-03-07:
Manual corrections won't work. It depends on clip axis of rotation (how distant to top of the spine inside the skull), camera angle, sitting position etc...

sthalik, 2018-03-07:
And here's how strong Z affects Y and pitch:

```
                 z
         x   0.497
         y  -0.508
         z   1.000
       yaw  -0.033
     pitch   0.216
      roll  -0.102
```

Len62, 2018-03-07:
I can see that this would be great when implemented. Probably of use for anyone whose set-up is even slightly off the horizontal/vertical plane.

Simon-Dainty, 2018-03-07:
I look forward to testing the, er, testing build.

sthalik, 2018-03-09:
Thank you for your encouragement. It actually means a lot. If not for opentrack's continued usage, and positive opinions, I would've thrown in the towel a long time ago. Developing the project is half fun, half a burden. 👍 🐙 

The last few bits of any new feature is making sure it works under supported operation, and receiving tons of (warranted) bug reports. At times I can't handle the stress, like now. The correlation calibrator was written third January.

I've tried making arrangements for a second core developer I could mentor toward understanding of opentrack internals, that could do a part of the same work I do now. Negotiations fell apart but I'll make other arrangements.

Despite that, I would never wish the project never occured. I wouldn't be the same person I am today, personally or even profesionally.

In case the project actually goes the way of the dodo, which isn't likely, I'd greatly appreciate it to be taken over by someone competent. There are several inactive users with administrator privileges who can in fact pass ownership directly.

**TL;DR** complaining about things being hard, work will go on, but slowly

closed 2020-01-27 by sthalik