opentrack/opentrack #755: [TrackHat] New Clip dimensions are incorrect

sanpats, 2018-02-27:
I just bought TrackHat Clip Plus Mini (the new clip in your code, trackhat branch). But my measurements are different than what you put in a recent commit. Mine are by = 62, bz = 35, ty = 43, tz = 14. You're better check with TrackHat.

sthalik, 2018-02-28:
I elevated the issue.

Now, my own questions. The clip I own wasn't made by trackhat, I only provided the size for their manufacture.

1. What kind of measuring device did you use?
2. What size is a single trackhat LED?
3. Did you get the depth values by Pythagoras?

sanpats, 2018-02-28:
1. A ruler and a square. Plus a sheet of paper to mark and out line the LEDs and the shape. 
2. I tried to measure from the center of the base as the height of the LEDs vary between the three LEDs +- 2mm (I think they were cut at the top and sanded down by hand). 
3. I drawed 3 parallel lines that pass through all the centerlines of all three LEDs, plotted the outline, and measured the distance between 3 LED bases.

sanpats, 2018-02-28:
My measurement may be error 1-2mm but should not be more than that, but as the y axes are 43 and 62 the same as older clip, I think it is intentional.

sanpats, 2018-03-08:
I read your reply amp n BMS forum that it is the old clip with z values cut in half. 
The track hat clip pros, both old and new, have the same y value, by = 62, ty = 43. 
The different are z values, the old one are bz = 75, tz = 27. 
The new one are bz = 35, tz = 14; which are approximately half of the old one. (The exact half of the old one would be 37.5 and 13.5 respectively.

sthalik, 2018-03-09:
New values:

> by = 60, bz = 38.2, ty = 42.2, tz = 12.6;

I'm releasing a TEST build.

sthalik, 2018-03-09:
Get it at http://opentrack.misaki.pl. You should know which file it is.

I'd expect trackhat to release an official version soon.

sthalik, 2018-03-09:
Please confirm whether the damn thing works.

sanpats, 2018-03-09:
I'll test the trackhat opentrack. What about normal opentrack? I cant input floating point, only integer. Will the nearest integer suffice?

sthalik, 2018-03-10:
Nearest will suffice.

sthalik, 2018-03-18:
I'm keeping this issue open until p6 comes out.

sanpats, 2018-03-20:
New dimensions work fine with p5, but I didn’t test it very extensively as I normally use normal opentrack with my Trackhat clip plus mini.

sthalik, 2018-03-20:
Use the test build or the new dimensions in opentrack. The issues manifest themselves in subtle ways. The algorithm is robust in face of incorrect values but there's some "residue". You'll also need less filtering with correct model.

sanpats, 2018-03-20:
Sorry I meant test build not p5. I tested in normal opentrack as well. Both works fine.

sthalik, 2018-05-14:
Complain loudly till it's fixed. I can only nag relevant people to a certain degree.

sthalik, 2018-05-14:
FWIW, I sent p6 already but it hasn't been put on the site.

sthalik, 2018-08-03:
They should release p7 soon. Thank you for reporting this.

closed 2018-08-03 by sthalik

sthalik, 2018-10-27:
@sanpats looks like version p7 is now on trackhat's site. Sorry for the belated reply.