opentrack/opentrack #756: Is it possible to use the oculus touch controlers?

Neuse87, 2018-03-07:
I tried the oculus hmd support an i works great as a normal headtracking device straped on the forehead.
Would it be possible to do the same with the oculus touch controlers? Like strap one to a headset and say you are my headtracker now?

sthalik, 2018-03-09:
Send me the link to the SDK download and documentation.

Animus777, 2018-04-24:
Not sure if still needed but I'll post links anyway:

**SDK Download** [https://developer.oculus.com/downloads/package/oculus-sdk-for-windows/](url)

**Documentation** [https://developer.oculus.com/documentation/pcsdk/latest/concepts/dg-input-touch/](url)

I am very interested in that feature. Mostly because it allows to use Oculus Touch as a Mouse with non VR games. 

It's definitely possible. Some guy from oculus forums is trying to achieve it via AutoHotkey [https://forums.oculusvr.com/community/discussion/comment/576525/#Comment_576525](url) But I woud prefer Opentrack solution.

sthalik, 2019-02-11:
I don't know what additional code is required for a touch controller. We're using the SDK version 1.4.0 and it's quite old. Updating the SDK requires additional work and I can't do it without someone to test a particular device, either.

closed 2019-02-11 by sthalik