opentrack/opentrack #758: Feature request - movement history with reltran

sanpats, 2018-03-08:
I don’t know if it’s implementable or appropriate, but having an option to turn on movement history for reltran like TIR’s TrueView feature would be a nice addition. Would be even better is to have easy UI option to enable such feature according to personal preference.

sthalik, 2018-03-09:
I agree this is a very useful feature.

However I have my own reservations against copying ideas bit-by-bit.

Let's let this issue remain open for the time being.

sanpats, 2018-03-09:
This is my idea:

Use something I would call incremental movement, just append the current movement without considering the movements that came before. This will result in similar fashion with Trueview.

(With reltran)
1. move toward monitor without any rotation -> increase z axis
2. yaw to 90 degree left -> decrease yaw axis (normal reltran will take the fact that you've already moved in z axis, and will increase yaw together with gradually translate z axis value to x axis. Incremental movement will ignore that previous movements and just yaw on the spot)
3. move toward monitor -> decrease x axis

This will not guarantee return to center.

Can be solved by set a center bubble as a threshold. Within this threshold, incremental will turn off.

sthalik, 2018-05-27:
I'm thinking of scaling position back to zero the more user returns to physical center. Whether linear scaling works I don't know, but it could be exponential scaling or something.

I won't be copying NaturalPoint while they're still in business but that's a very useful feature and let's keep it open.

sanpats, 2018-05-28:
No need to copy them. I think there are many different approaches to a similar feature.