opentrack/opentrack #759: need someone to step in as QA

sthalik, 2018-03-09:
Hey,

Programmer's job is normally is separate from QA. The latter is a tester, both for normal usage scenarios as well as obscure ones the programmer forgot about. Right now I have both responsibilities.

Unfortunately there are requirements:

- **none** previous professional QA experience needed!!!
- a tracking device, using it ingame
- instant messaging realtime contact
- voice over Skype is fine too (decent understanding of English welcome)
- availability at least every 2-3 days for 1:30 to 2 hours, when many changes are sent in a burst (only when programming work is actually done at all)
- willingness to learn Git, where I provide mentorship
- checking newly-made code under both easy and extreme scenarios
- willingness to install Visual Studio for debugging and compiling
- willingness to learn building the project by themselves, where I mentor (has been done before, not hard).
- taking notes for procedures like building or Git usage

Benefits? Unfortunately they aren't very substantial. The singular one is visibility in credit list and each release text. The other one is havings builds before any other user.

This is in no way a permanent position and you can of course quit at your leisure if time forbids or the task is too tedious.

I'm paging several users now. Please note that this is **NOT** guilt-tripping and you're in no way obliged to help. Paging @Len62, @MathijsG, @mrsanchos, @albertolg.

Len62, 2018-03-10:
Give me a couple of days to think this over. Sounds interesting, I just don't know if I have the drive to stick with it.

MathijsG, 2018-03-12:
Having the same thought. Sounds something I could do, already know git, although the frequency sounds somewhat high, I hope to graduate soon and dive in some Java traineeship, so maybe I can try some easy programming parts later on.

At one side I like to be part of this active development and on the other side I think "what's currently not working now?" I think many of our users never even updated because their DIY soldered headtracking did worked fine for them already with a much earlier version.

Do you have some roadmap in your mind about where you want to head with the program?

sthalik, 2018-03-12:
@MathijsG the frequency is only important when there's high commit activity. That is, the worst-case scenario. I'd expect QA to respond within 1-2 days when there's something big, or a period of unsuccessful fixes for the same problem.

You can check the "commit log" on the project's main page, click "[number] commits". Normally there's not much going on, but I'm talking about the absolute worst-cases (big change, fix attempts).

sthalik, 2018-05-14:
Such sad :(

closed 2018-05-14 by sthalik

MathijsG, 2018-05-14:
Sorry man, currently I'm too busy with graduating that I can't fulfill this in a proper way. When this is over and I have more time I would gladly help.