opentrack/opentrack #760: FSX simconnect error

TommyLee42, 2018-03-09:
I needed to reinstall opentrack (wich was working before on my FSX:SE (win 8.1 64 bit) but now I get this error:

[](https://imgur.com/a/u5d3X)

sthalik, 2018-03-10:
Install FSX simconnect SDK. It's necessary for any operation. Not our fault, stupid Microsoft felt that way.

TommyLee42, 2018-03-10:
I did installed that from FSX:SE SDK folder..all three of those long time ago, I even went for checking installation and repaired all three simconnect SDk version, but still not working. I forgot to mention that.

TommyLee42, 2018-03-10:
I went ahead and checked windows/SxS folder and found alot of simconnect files there. Is this good or not, I am not sure but i am a bit stuck now. Funny, everything else works..Btw i am using opentrack 2.3.10 on 64 bit windows...

sthalik, 2018-03-10:
You may need to install updates to the SDK. The process is quite tedious.

sthalik, 2018-03-10:
OR, you may install the oldest SDK version bundled with your FSX copy.

pgazda, 2018-03-15:
I had same problem. Reinstalled Opentrack after some time and got same error message. In the end I solved it by going back to version 2.3.9.

sthalik, 2018-03-16:
There indeed appears to be an issue with loading SxS manifests in opentrack's simconnect module. It's very minor and absurd. Breaks FSX, nonetheless.

I don't have it installed but this one line change should fix it. See the `20180316_01` build -- <<http://opentrack.misaki.pl>>. Please confirm whether it works.

pgazda, 2018-03-16:
Thanks for quick reply. Works with build `20180316_01`.

TommyLee42, 2018-03-19:
Redownloaded and reilstalled opentrack in meantime and no error this time.
Must be new build that I installed or something..dunno

sthalik, 2018-03-19:
I'm releasing a new build due to this issue, shortly.

sthalik, 2018-03-31:
@TommyLee42 I can assist more given more detailed info.

LTRMcrew, 2019-09-22:
> There indeed appears to be an issue with loading SxS manifests in opentrack's simconnect module. It's very minor and absurd. Breaks FSX, nonetheless.
>
> I don't have it installed but this one line [...]

Can confirm this fixes the simconnect issue with FSX. Thanks for the work.

SargeT, 2021-01-19:
Still not working in 2.3.12.
[image]
Tried every option from here:
[image]
Tried solution from #841 but couldn't find libopentrack-proto-simconnect.dll to rename.
[image]