opentrack/opentrack #762: freetrack: registry "NPClient Location" overwritten on start

ItGalactic, 2018-03-17:
When using Output "freetrack 2.0 Enhanced" with "Use FreeTrack, hide TrackIR" option,
the registry "NPClient Location" is overwritten with blank on every re-start.

And then will need to use the "Repair NPClient Location"/"Locate DLL" button to re-set this registry 
while the tracking has started.

OS: Win 7

sthalik, 2018-03-17:
From your description it works as intended. What is your issue?

ItGalactic, 2018-03-19:
"NPClient Location" registry = blank makes some programs not work.

In comparison, if "Enable both" or "Use TrackIR, hide FreeTrack" option is chosen instead, then
on restart of tracking, that registry is overwritten with the opentrack programs "modules" directory,
which make it work.  So having been set with blank seems to be an error?

sthalik, 2018-03-19:
It's set to blank for the case of games that support both protocols. Having both head tracking sources makes them confuse themselves.

ItGalactic, 2018-03-19:
I see, so I think you are saying "NPClient Location" registry is only used by TrackIR, not by FreeTrack.

But, then,  looks like the game Elite: Dangerous 3.0.3 has a bug in that it requires that registry
to be set in order for FreeTrack protocol to work.

Thanks for your help.

sthalik, 2018-03-31:
@ItGalactic you need the "TrackIR" protocol enabled for ED to work at all.

ItGalactic, 2018-03-31:
@Stanisław, 
        Maybe my case is special or something, but my ED setup works
    with FreeTrack protocol.
    My OS is Win 7 x64 SP1, and I do the following steps:  
    
    1) Set to "Use FreeTrack, Hide TrackIR" in "freetrack 2.0 Enhanced"
    options.
    2) Start Tracking
    3) Repair NPClient location using the "Locate DLL" button in
    "freetrack 2.0 Enhanced" options
        and set it to the opentrack\modules\    directory.
    4) Start ED
    
    
    On 2018-03-30 18:03, Stanisław Halik wrote:
    
      @ItGalactic
        you need the "TrackIR" protocol enabled for ED to work at all.

sthalik, 2018-03-31:
@ItGalactic just set protocols enabled to "Enable both" or "Enable trackir". You're doing exactly what the "Enable both" setting does.

ItGalactic, 2018-04-02:
@Stanisław    Steps I outlined seems to be different than "Enable
    both"
        since the TrackIR process (as shown using Task Manager) does not
    startup
        for the steps I used.
    
    
    On 2018-03-31 07:53, Stanisław Halik
      wrote:
    
    
      @ItGalactic
        just set protocols enabled to "Enable both" or "Enable trackir".
        You're doing exactly what the "Enable both" setting does.

sthalik, 2018-04-02:
It's only there because few games don't work with head tracking otherwise.

Please fix the formatting, see how your posts look on the website.

sthalik, 2018-04-02:
And just to be very clear, the process doesn't do anything. It only sits doing nothing.