opentrack/opentrack #764: Issues with Build Guide/Installing Dependencies

babbott1000, 2018-03-25:
I apologize, as I have not even heard of most of the dependencies that are required to build opentrack, and much less used(or installed) them.  I also do not use Visual Studio(and C++) often.  I had tried a while ago to try to build opentrack but without success.  This would be more considered a feature than an issue as I understand that there are more important issues that need to be worked on, but the [build guide](https://github.com/opentrack/opentrack/wiki/Building-under-MS-Visual-C---2017-and-later) is really confusing and there is a lot of conflicting information.  I would appreciate any improvements anyone could make to make it easier to understand.

sthalik, 2018-03-26:
Which compiler are you using then? The guide is written specifically for Windows users, but the mingw-w64 toolchain on Windows is also explicitly supported.

babbott1000, 2018-03-28:
For example in the qt instructions it says “Use the .zip file from http://download.qt.io/” but the link http://download.qt.io/ gives several files to download when opened and it is not specified which one it is talking about.  Another example is right under it there is a header titled partial qt build and it is not mentioned wheather you have to do that in conjunction with or in replacement of the instructions with the header “Installing Qt built with MSVC++ 17 Preview”

sthalik, 2018-03-29:
@babbott1000 are you a programmer?

babbott1000, 2018-04-02:
I have programmed in several lanuages including python, nodejs, unity c#, and made webpages with html JavaScript and css(all of this is on my free time and because I enjoy it, not professionally for a living ).  I have no experience with visual studio(except for using it as a simple IDE) and installing dependencies for c++ and other similar languages.  I understand that this is project(open track) is made only by the community and there may be errors in the wiki, but I could not understand any of it, not because it was hard to understand but just because it was downright confusing.

sthalik, 2018-04-15:
@babbott1000 
1) Build it on Linux first (within, say, WSL). It's a lot easier.
2) The instructions make sense but knowledge of building C or C++ projects is presupposed.

babbott1000, 2018-04-15:
@sthalik Is there a way to build it on windows, could install Linux on a flash drive, but It would take forever, and I don't have any space on my HD

sthalik, 2018-04-15:
Build in a VM or WSL.

babbott1000, 2018-04-15:
@sthalik Are there any binaries or do you have to build it?  I don't understand why there would be an installer if you have to build it.

sthalik, 2018-04-15:
I build the Qt version for opentrack when it's updated. Same for opencv and the rest of open libraries.

closed 2018-07-13 by sthalik