opentrack/opentrack #765: Relative Translation Change

Panzerm4n, 2018-04-01:
Just updated from 2.3.1. There has been a change to the way relative translation works. When I translate in the X direction and then yaw, my x values are reversed at any yaw value above around |90 deg| (output). I only need relative translation around |170 deg| yaw. So
1) Is this correct or is there a problem with my calibration?
2) Is there a way of modifying the angle at which relative translation occurs?

I play WW2 sims and only need relative translation to look over my shoulder. I want to lean out of the cockpit on takeoff and look around, and relative translation is pushing me back into the cockpit when I yaw.
Thanks

Len62, 2018-04-01:
I take it you are on 2.3.10 now, and did you carry over your ini file? First thing I would try is to load a new blank configuration, re-calibrate, and adjust the other settings. In IL2-BOS I can lean out of the cockpit and yaw without any problems. This is with relative translation disabled.

Panzerm4n, 2018-04-01:
Yes, I'm on 2.3.10. It's fine with relative translation disabled. But i do want it enabled to check my six. I'm playing BOS as well.

sthalik, 2018-04-01:
Which of the two "enabled" modes are you using?

Panzerm4n, 2018-04-02:
I've tried both "enabled" and "enabled when not aiming." They both produce the same result

Len62, 2018-04-02:
Okay, I enabled relative translation and I'm seeing the same issue. I will make a short video in the morning and post it here. I always kept relative translation disabled which is why I was not seeing it.

sthalik, 2018-04-02:
@Panzerm4n @Len62 

Disable dynamic neck for testing if you have it. It may cause that effect.
I set my yaw to 0. Having some values for X. Now:

- having +-180 yaw reverses yaw
- having +-90 yaw moves X to Z

sthalik, 2018-04-02:
Just to be clear my X is zero on yaw 90, with no Z input.

sthalik, 2018-04-02:
I did my testing on the build from http://opentrack.misaki.pl. That is, the newest unstable build. Are there any differences between it and the stable build?

Len62, 2018-04-02:
The stable 2.3.10 build only impacts the X movement, the test build gives me the movement along the Z axis also. Video should be done shortly, I'll do both versions of OT.

sthalik, 2018-04-02:
@Len62 please try to do this:

- disable dynamic neck
- enable reltrans unconditionally
- use a test build
- set Y and Z for "relative translation only"
- try +-90 and +-180

The results should be the same for the released build.

Len62, 2018-04-02:
Ok, enabling only Y and Z for reltrans makes it work correctly in the latest test build and the current release build.

sthalik, 2018-04-02:
Sorry, that's not what I meant. Keep X at enabled.

Len62, 2018-04-02:
Enabling X did not change anything in either version.

sthalik, 2018-04-02:
Show a video please.

Len62, 2018-04-02:
Okay, give me a little time to capture one.

sthalik, 2018-04-14:
@Len62 keep everything you want at enabled, both in `options -> output` and `options -> reltrans`. I need to see you problem in action.

Len62, 2018-04-15:
@sthalik  Link for this issue: https://www.dropbox.com/s/wdsuoi7kfysqkjk/Xtransissue.mp4?dl=0

sthalik, 2018-04-16:
@Len62 I don't see any geometry issue. It may be easier to understand the behavior if you rotate first, then move.
@Panzerm4n use the "enable when not aiming" reltrans mode. When in doubt, use a build from <<http://opentrack.misaki.pl>>.

Len62, 2018-04-16:
I did some more looking at how the movement works with reltrans on and have a better grasp of the geometry. I don't use reltrans normally, just got curious after reading the OP.

Len62, 2018-04-18:
No difference.

Panzerm4n, 2018-04-19:
This is really bugging me. It severely affects tracking a target. Here's a video of the test build with all three 
reltrans options selected. Selecting "not aiming" only helps to a certain degree. It does not fix the problem. 
https://youtu.be/FKcsPvFcWSA

Panzerm4n, 2018-04-19:
Ok, but the whole reason WWII simmers want relative translation is to lean left and right while they look behind them... I hope you consider adding the option where reltrans is enabled only at extreme yaw values. Guess it's back to an older build for now.

FlyMakerBR, 2019-10-09:
Please review this issue.
We use relative translation to apply X after Rotation... No need Z input for that
Can we have a option for this?

sthalik, 2019-10-12:
It's not a trivial change. I'd have to throw away all reltrans code and write something five times longer. Too bad the request isn't accompanied by code.

FlyMakerBR, 2019-10-12:
I found a workaround

Set X axis to Yaw, put deadzone in X until all yaw range was reached, start X translation after that. Working nicely for me in the last beta version

closed 2020-01-27 by sthalik