opentrack/opentrack #768: Roll Translation Reverses after Yaw 180

Len62, 2018-04-13:
So, looking straight ahead and rolling left/right the output is correct. However, when yawing 180, the roll movement becomes inverted. I've tried w/all the options with reltrans enabled, and disabled, with no change in the movement. I don't normally keep roll enabled but I was testing because it seems other users are experiencing the same issue. Video to follow in a little while.

sthalik, 2018-04-14:
A video will help a great deal!

@Len62 github doesn't send notifications for edited messages. When sending the next video, please make a separate message.

Len62, 2018-04-15:
@sthalik  Here's the link: https://www.dropbox.com/s/c1918nqs72swa7o/rollreverse.mp4?dl=0

Maetharin, 2018-04-15:
I don’t really see the problem? Isn’t that how rel trans is supposed to work?

Or is it that it doesn’t stop to do that even when RT is switched off?

Len62, 2018-04-15:
That was w/reltrans off. I don't normally use the roll axis, so I never noticed it. I was testing as some other users were reporting the issue on another forum.