opentrack/opentrack #770: View jumps with 2.3.10 accela Aruco

Emton, 2018-04-19:
Hey just letting you know I tried 2.3.10 and I noticed the view gives these sporadic jumps when using accela/aruco even just looking forward. I go back to 2.3.9 and the view holds steady no problem.

Psebcool, 2018-04-27:
Same issue.
I have a screen jumps with OpenTrack 2.3.10. 
I don't know if it's Accela's fault. I use Point Tracker 1.1, Freetrack 2.0 enhanced and Accela. 
I tried FaceTrackNoIR with PointTracker 1.1, Accela MK2 and FreeTrack 2.1, and no view jumps. perfect.
I will try with OpenTrack 2.3.9.

JoeSchr, 2018-05-09:
omg, going back mages it so much better. thanks for reporting this

sthalik, 2018-05-14:
In order to diagnose whether the issue lies within Aruco or Accela, I need you to do some tests.

For starters, increase Accela deadzone to the max value and see if it still happens. Come back with the result.

Emton, 2018-05-17:
Ok I turned the deadzone up to max and didn't notice the jumps

sthalik, 2018-05-17:
What was your deadzone value in 2.3.9 and what's the lowest one that's comfortable now?

sthalik, 2018-05-17:
See if today's third build makes it any better.

Emton, 2018-05-17:
Is that in your dropbox could you give the link I forgot it

sthalik, 2018-05-17:
http://opentrack.misaki.pl

Emton, 2018-05-19:
In the build opentrack-test-20180517_03.7z the Octopus doesn't move at all, if I hit the toggle key then it starts flipping around with crazy pitch values of like 9000+

sthalik, 2018-05-21:
Is it a particular filter or ANY filter?

Emton, 2018-05-21:
I tried the other filters and they would move very very very slowly but the accela filter was frozen and wouldn't move at all

sthalik, 2018-05-22:
Please try today's build.

Emton, 2018-05-24:
Ok I tried that build it seemed to work pretty good no major jumps I'll try some more later using my usual lighting conditions too

sthalik, 2018-05-25:
See if it's any worse than 2.3.9. It's meant to be no worse.

sthalik, 2018-06-28:
See #778

closed 2018-06-28 by sthalik