opentrack/opentrack #771: Build of tracke-steamvr fails in linux with conflicting declaration of "error_t"

zanoni-mbdyn, 2018-04-27:
Hi everybody, 
i tried to build opentrack in linux (Arch and Ubuntu 16.04) enabling the module tracker-steamvr. The build iniitially failed though, with this error:
```
In file included from /home/zanoni/lavoro/simulatore/vr/opentrack/tracker-steamvr/steamvr.cpp:19:0:
/home/zanoni/lavoro/simulatore/vr/opentrack/tracker-steamvr/steamvr.hpp:20:33: error: conflicting declaration ‘using error_t = enum vr::EVRInitError’
 using error_t = vr::EVRInitError;
```
I had to manually rename every occurrence of `error_t` in `steamvr.cpp` and `steamvr.hpp` to `vr::EVRInitError` to make it work. Did I miss something, maybe in the openvr build or in the cmake configuration, or this is a bug? 

Cheers

sthalik, 2018-04-27:
What's the compiler version?

zanoni-mbdyn, 2018-04-28:
It's gcc-7.3.1 in both cases

zanoni-mbdyn, 2018-05-02:
OK, thanks. I'm trying to use opentrack to use the HTC Vive with FlightGear in a research project. 
I'll keep you updated with our progress.

zanoni-mbdyn, 2018-05-02:
Just tried after pulling the last snapshot. Same error with `error_t` as before, I still had to manually substitute all of its occurrences to `vr::EVRInitError` to make it compile. 

When I click on `Start`, after selecting `Valve SteamVR` as input and `FlightGear` as output, this exception is launched: 
```
VRInitError_init_VRClientDLLNotFound
```
can you please help me understand what is going wrong (or I'm doing wrong) here?

Thanks

P.S. To complete the build I also had to add this include in `migration/migration.hpp`
```
#include <memory>
```
otherwise, gcc complained that `shared_ptr` is not a member of `std`

sthalik, 2018-05-02:
For `error_t` the compiler shows places of both the original definition and the redefinition. Please paste these.

sthalik, 2018-05-02:
As for your missing shared library error, that's an issue with your local configuration. Use `strace(1)`, read steamvr docs, or both.

zanoni-mbdyn, 2018-05-03:
You can find the whole output of the compiler in the attached file. And thanks for the tip on the missing shared library, I'll try to explore a bit. 

[tracker_steamvr_build.txt](https://github.com/opentrack/opentrack/files/1970833/tracker_steamvr_build.txt)

sthalik, 2018-05-03:
@zanoni-mbdyn you should be able to build now. GNU libc added an idiotic typedef so it didn't build.

sthalik, 2018-05-03:
Please confirm if it builds on Linux now.

zanoni-mbdyn, 2018-05-03:
Yes it does. Thanks.

closed 2018-05-03 by sthalik