opentrack/opentrack #773: Opentrack doesn't receive data from FreePIE IMU but receives from another app

StuDentBRz, 2018-04-30:
I am trying to use opentrack with FreePie IMU, on my android smartphone with gyroscope and accelerometer , but i am having some trouble...
I can see on FreePIE IMU on Android that it is getting the acelerometer and gyroscope data, but opentrack won't receive any of this data. I've tried via wi-fi network and USB tether, i always configured the same IP and ports, and even disabled windows firewall and nothing worked.
It's not a problem on my network, because i've tested with another app on android (HeadTracker from play store) and also changed the input from FreePIE UDP receiver to UDP over network and then opentrack received the data as it should...

sthalik, 2018-04-30:
I need to see the packet payload. Are you able to use Wireshark?

sthalik, 2018-04-30:
First check if version from opentrack install directory's `doc/contrib/freepie-udp` works for you.

sthalik, 2018-05-01:
Disable Windows firewall temporarily, start opentrack and see if it works. If it does, it's a firewall problem. Also check port and IP address.

sthalik, 2018-05-01:
Oh, and also confirm that the phone has all of: accelerometer, gyroscope, magnetometer.

StuDentBRz, 2018-05-01:
My smartphone is a Motorola moto G4 plus (Android 7.0), It has gyroscope and accelerometer but no magnetometer. As i Said in the first post i already tried to disable firewall and checked the ports, i even opened the 5555 port on firewall myself and didn't work.
I thought It was a because it didn't have magnetometer, but It Works Just as It should with the other app (headtracker).
Unfortunatelly i don't have a Phone with magnetometer to see If It Will work...

sthalik, 2018-05-01:
In the Android app, select "send orientation", check "debug", and see if you get any rotation data after starting.

StuDentBRz, 2018-05-01:
No i don't get any orietation data, i can only get data when RAW input is active and It is only gyroscope and acelerometer, no magnetometer or orietation.

sthalik, 2018-05-01:
`opentrack` specifically asks for rotation data. If you select "send raw" on Android but debug shows no rotation data, the phone isn't supported.

sthalik, 2018-05-01:
Just to be clear, please note that rotation data is extracted using Android's internal API. We don't implement the estimator.

sthalik, 2018-05-01:
I forgot again, apologies. I've tried extracting data using only gyro+acc, it had serious problems and wasn't really usable.

Consider using the Aruco tracker (info on github's wiki) for a zero-cost solution.

closed 2018-05-01 by sthalik

StuDentBRz, 2018-05-01:
It's really weird that you say that my Phone is not supported  by opentrack, because it Works Very well Just by using headtracker app(which i downloaded from Playstore), instead of FreePie IMU,  i am just going to keep using this way... It is Free with ads, but it doesn't really matter for me, because i won't be looking at the app for much longer... At least It is a option for people like me who has a Phone with gyroscope and acelerometer but no magnetometer, so i can have opentrack working without have to buy a New Phone... Did You try to use Headtracker to see what is the difference the data difference between Headtracker and FreePie IMU ?

sthalik, 2018-05-01:
There's an AHRS method for using gyro+acc only but we don't implement any methods on our own in the .apk.

diveyez, 2019-01-13:
The FREEPIE apk is not sending any data to windows at all, entirely broken UDP code

sthalik, 2019-01-14:
@diveyez the code's not broken. The apk's been like that for a long time and hasn't had problems sending data. Did you check with Wireshark?

dorianr666, 2019-04-23:
yep, its broken. looks like freepie imu doesnt send data. other apps communicating with opentrack from android work fine.

sthalik, 2019-04-23:
What's the Android version?

MABlosfeld, 2021-11-14:
freepie imu is working but when i turn on another app or game, freepie turns off by itself. i'm using opentrack/moonlight/reshade with my vr headset and android, just need freepie to transmit data from motion sensors to opentrack. I support paying a small amount to keep all these projects and developers motivated.

sthalik, 2021-11-16:
There are several .apk versions in the contrib directory. One of them was known to work with Android 9.

MABlosfeld, 2021-11-16:
I'm using "freepie_imu_android9.apk" apk and the app is working, the problem is that when I start moonlight or any game, freepie shuts down by itself. My android is on version 12, maybe this is the reason that makes freepie turn off by itself.

MABlosfeld, 2021-11-19:
Moonlight and other games are added within the GAME LAUNCHER APP which has a priority mode, I disabled priority mode and now FREEPIE is not turned off when I start moonlight. Priority mode is important, it keeps the screen always on. Now I need to find a way to add FREEPIE inside the GAME LAUCHER app.

MABlosfeld, 2021-11-21:
I disabled priority mode in game launcher, started freepie and then started moonlight. After 20 min everything was still up and running (freepie/moonlight/cell phone display).