opentrack/opentrack #774: P3D view Group issue

503Kento, 2018-05-01:
I recently created a view group for two of my three monitors.  Opentrack works as expected until I click on the secondary monitor of the view group. 

Tracking remains on and if I switch to the octopus, I see the IR LED’s in the camera window and the game data numbers changing. It’s as though P3D is no longer “connected” to opentrack. Once I click back on the primary monitor, tracking begins functioning again. 

I’m not 100% sure this is an opentrack issue as it seems like it cou be a simconnect issue or a “feature” of view groups but I’m not sure how I could tell where the issue lies. 

Thanks for your help,
Kent

sthalik, 2018-05-02:
You can run `opentrack.exe` from Windows Command Prompt. It'll show messages about being disconnected from `simconnect` if that's the case.

Other than that, opentrack doesn't care about simconnect monitor count.

503Kento, 2018-05-03:
Did as you suggested, it looks like simconnect is still connected so I’m not sure where the problem is.

sthalik, 2018-05-03:
It's in P3D. There's nothing opentrack can do from its side.

Maybe it'll help if you used the option to connect all displays into one area.

closed 2018-05-14 by sthalik