opentrack/opentrack #775: Fusion of FreePIE amd Aruco crashes in 2.3.10

Artyrm, 2018-05-01:
Separately they both work fine.

I managed to debug it in VC2017, got this:

[image]

Artyrm, 2018-05-02:
Interestingly enough, if I change their order, i.e. Aruco for rotation and FreePIE for position (which is of no real use, of course), all works fine.

[image]

sthalik, 2018-05-02:
This is an obvious-to-fix bug. Stay tuned.

sthalik, 2018-05-02:
See the build at <<http://opentrack.misaki.pl>>. There may be some new regresions since I haven't given the new commits enough testing. Please confirm if it works.

Artyrm, 2018-05-02:
Yes, this bug is completely gone.
Thank you very much for what you're doing.

Artyrm, 2018-05-02:
If it matters, in this late build axes Pitch and Roll are completely messed up for me in FreePie.

sthalik, 2018-05-02:
What's are the raw values for pitch and roll on the main window?

Artyrm, 2018-05-02:
Should we start a separate thread on this? 

Anyways, I found this convenient, that video from aruco can be shown now alongside with octopus movement, so I recorded my window:
https://www.youtube.com/watch?v=ahU8zhb0nng

Single FreePIE tracking behaves exactly the same.
So maybe my "Pitch and Roll" statement is an underestimation here.

In 2.3.10 all I had to do is little remapping and all went almost fine (except some jumps at limit points like 90°). 
[image]

Here it doesn't help at all.

Artyrm, 2018-05-02:
I did. Also I had to remap the same axis, and invert one (this time Roll instead of Pitch).

It got almost normal, but something is still fishy with Pitch, while I tried to keep it well away "straight down, straight up" positions (what's with 90° anyways, on phone VR apps there are seems no problem with that?). 
https://www.youtube.com/watch?v=lLaraz_M74A

What bothers me the most it is inconsistency with the settings of 2.3.10 - do they really had to change here? I mean, those are axis mapping and default corrections, there seems like little point to alter them in software?

sthalik, 2018-05-02:
What inconsistency are you talking about?

Artyrm, 2018-05-02:
And there seems to be the problem with Fusion + aruco axis as well.
First I tuned up aruco - changed X and Y axis (strangely enough). 
[image]

But when I switched to Fusion, obviously axis become messed up. Here is the video:
https://www.youtube.com/watch?v=l9uv04RE-DM

Are you sure you don't want us to start a new thread like "Axis problem with Fusion"? I can move my posts from here like no prob.

Artyrm, 2018-05-02:
> What inconsistency are you talking about?

I mean, I had no correction for roll in 2.3.10 and the other one axis inverted  (Pitch instead of Roll). But in the new build I had to change those settings - that means that profiles will need to be corrected for everyone in the new build?

sthalik, 2018-05-02:
You need to adjust the "add to axis" values each time you reorient the device around the camera. The roll value isn't valid.

> I had no correction for roll in 2.3.10 and the other one axis inverted (Pitch instead of Roll). But in the new build I had to change those settings

By "correction" do you mean having to change the raw value? If so–there's a bigger problem with how gimbal lock was worked around, and that was fixed in turn bringing in current behavior. I chose roll as the gimbal lock-inducing axis of rotation as it's used the least.

Artyrm, 2018-05-02:
Yes, I mean that in 2.3.10 I had working settings for FreePie (that glitched a little only at "Round values"). In this build we had to alter these settings and still, there is a problem with Pitch axis. Here are my settings for 2.3.10 (left) and for the current build (right):

[image] [image]

[image]  [image]

>If so–there's a bigger problem with how gimbal lock was worked around
That's what I'm afraid. What could be the plan here?

I'm sorry, I should go get some sleep, I can answer this time (minus couple of hours) tomorrow.

Artyrm, 2018-05-02:
I'll try this.

Avtually, I was wrong about "No problem" with 2.3.10. It's just harder no notice. You can see there is also problem with the pitch movement. Notice how output Roll changes while input Pitch (there is Pitch->Roll translation in my config) stands still:
https://www.youtube.com/watch?v=otVn0o7dzmM

Gone till tomorrow.

closed 2018-07-13 by sthalik