opentrack/opentrack #776: Opentrack stopped working as soon as PS3 camera get's used

MathijsG, 2018-05-02:
Today I tried opentrack with the PS3 camera at a system with Windows 10, latest version and all and with the CL-Eye drivers 5.3. But unfortunately whenever I tried to start the tracking I get this window:
**
[image]
**

And without further notice the program kicks out. When I try a different camera attached to the PC Opentrack doesn't kick out.

I tried:

- Other versions of opentrack
- FaceTrackNoIR

and they all behave the same.

When I look at the PS3 Eye image in CL-Eye test program I do get proper image without errors.

Is there some way I can log this behavior more in-depth? The error screen shown above doesn't supply any info. I know this is just partly, or maybe not even partly, related to opentrack itself.

I also scanned the installed software and removed the K-Lite codec pack, as I remembered there once was an issue about this and that being the cause, however after this uninstall it's still the same in this case :(.

Any directions or tips?

sthalik, 2018-05-02:
1. try disable antivirus
2. Replug the webcam into a different USB port
3. In the worst case, lower FPS to 30

> Is there some way I can log this behavior more in-depth? The error screen shown above doesn't supply any info

It's the driver's fault that crashes the entire program. There's nothing to display on opentrack's end.

MathijsG, 2018-05-02:
Will dive into it later. I now 'fixed' it by using Manycam as an in-between. It seems weird to me that CL-eye test and Manycam doesn't have a problem with the video feed, while all the tracking softwares do.

closed 2018-05-14 by sthalik