opentrack/opentrack #778: opentrack 2.3.11 release process

sthalik, 2018-05-04:
I'm gonna be releasing `opentrack 2.3.11` relatively soon.

Here's a list of features:

- [ ] fixing `Aruco` #770  (@Emton, @Psebcool)
- [x] keep the right modules if changing language
- [x] keep `Fusion` from crashing #775 (@Artyrm)
- [x] fix switching `-180` `<=>` `+180` but add gimbal lock on raw Roll value (Yuri Vostrenkov)
- [x] `SimConnect` was completely broken #760 (@TommyLee42, @pgazda)
- [x] add a more complete Chinese translation -- done by @cpuwolf 
- [ ] more strings can be translated (translations haven't caught up yet)
- [x] closing with tray enabled had problems (Yuri Vostrenkov)
- [x] better user interface for the `Wii tracker` -- done by @cpuwolf 
- [x] new `steamvr` SDK
- [x] make `reltrans not aiming` less strict
- [x] make `reltrans not aiming` also disable dynamic neck
- [x] make `reltrans not aiming` have a smoother transition (HOPPING_PONY on Russian IL-2 forum)
- [x] center reltrans when center button pressed
- [x] center is now a held button (CAHEK on Russian IL-2 forum)
- [x] set game name to unknown if it's not named
- [x] update compiler for more speed (theoretically speaking)
- [ ] `toggle while held` must stop filter output #777 (@r8d)

Maybe:

- [ ] finish and add gui for `module support`
- [ ] plug in `correlation calibrator` to PT (must decide on axis interconnect; widget is now mostly done)

Stuff no one cares about:

- [x] **fix Linux build in many places**
- [x] fix build with `GNU Make` on Linux and mingw-w64
- [x] fix build with `MSVC` after i18n changes (@cpuwolf)
- [x] fix `lrelease` class name conflicts in UDP tracker/proto
- [x] switch to C++17 more fully (thanks to new MSVC)
- [x] work with `opencv 4.0` prerelease
- [x] use newer `opencv 3+` APIs (it's good that 4.0 removed the old cruft)
- [x] migrations are now easier to write because thunks
- [x] ensure building the right `Aruco` version through a compile-test
- [x] don't include `Aruco headers` with opentrack
- [x] make code cleaner and shorter sometimes

sthalik, 2018-05-14:
I'm not aware of any ppa's. If someone writes the build scripts, I can maintain a PPA on my own. I won't write the initial `.deb` scripts since they're so complex.

`linuxtrack` supports different devices. You may want to use it.

Len62, 2018-05-18:
@sthalik With the 20180517 build (not listed any longer) I'm getting roll input while yawing. Haven't found a way to correct/adjust for it. Also, the 20180517_03 build does not produce any output. Tracker works, calibration works, but no output.

sthalik, 2018-05-22:
It's always changing what kind of interconnect we have (yaw with pitch, yaw with roll) when it's necessary to change the rotation order. It'll only get "fixed" to a sufficient extent once I'm finished with the correlation calibrator.

Len62, 2018-05-22:
@sthalik Same issue with today's build. I can do a recording if you would like. Started with a clean ini and set-up everything as I normally do.

sthalik, 2018-05-22:
@Len62 can you send me just one or two lines from the .csv log?

sthalik, 2018-05-22:
wow...

sthalik, 2018-05-22:
this is stupid...

sthalik, 2018-05-22:
@Len62 just try pressing the "hold" button for testing in the old build. Disregard the .csv stuff.

Len62, 2018-05-22:
@sthalik  Using "hold" doesn't make any difference. I tried with the 3/16/18 build and today's.

sthalik, 2018-05-22:
@Len62 I wasn't able to reproduce with either Falcon BMS or the bundled freetrack tester. Can you enable freetrack and see if `doc/contrib/freetracktest` returns any data?

Len62, 2018-05-23:
For output I always use freetrack. Running opentrack and running freetrack test, freetracktest is showing movement on the roll axis while yawing only.

sthalik, 2018-05-23:
How does it show when you look left/right in the program? Does freetracktest show it differently?

The second question is whether the data is transmitted correctly via freetrack.

sthalik, 2018-05-23:
Please send me a short video since it's hard for me to understand.

Len62, 2018-05-23:
@sthalik This is with the latest test version. Roll, pitch, and yaw mapping values have input deadspace of 3. You can see as I start to yaw that roll movement starts. However, you can see in the clip that I can roll w/o any movement being picked-up on the yaw axis. https://www.dropbox.com/s/s8nchwjt0etrkj6/2018-05-23%2004-15-35.mp4?dl=0

sthalik, 2018-05-23:
Send me screenshots for yaw and roll mappings. You can just ctrl+v into this text box.

Len62, 2018-05-23:
[image]
[image]

sthalik, 2018-06-09:
@Len62 some background:

There was a centering bug when the center position was close to +-180 of any axis. I had to abandon the previous gimbal lock workaround. The reason this happens is because order of rotations changed. The gimbal lock will now only occur with the roll axis. That's not a problem given it's impractical to roll 90 degrees even with an HMD. The same rotation is now interpreted differently in angles. That's unavoidable given that games request Euler (or rather, Tait-Bryan) angles.

I can't consider that small raw roll difference as a bug/feature. It's interesting why you need that much roll to begin with. I found anything over `raw=45 -> out=60` excessive even with the previous rotation order.

Len62, 2018-06-09:
I probably don't need that much roll like you said, I just never got around to playing with the raw/out numbers. As I said earlier I typically disable roll so it's not a big deal. I'll experiment some more. Thanks for all your support and efforts with Opentrack!

sthalik, 2018-06-20:
Do you get any issues in the newest build `opentrack-test-20180603_01`? As always the URL is <<http://opentrack.misaki.pl>>.

I'd like to proceed further with the changes but need an affirmative answer from 2-3 regular users. Please post the trackers/protocols you use normally.

Len62, 2018-06-20:
I've been using that build for a while now, other then the roll question from earlier, the build works fine. I keep the roll axis disabled. Using PT 1.1, freetrack 2.0 enhanced, and the accela filer.

sthalik, 2018-06-20:
How bad is it with roll if you set it to 1:1?

Len62, 2018-06-21:
It's the same. It's strange because I can roll w/o picking up any yaw/pitch movement (or at least a minimal amount), it's when I'm just yawing that the roll movement is being picked-up.

Len62, 2018-06-21:
Straight up if I understand your question. I'm going to build a new clip later on today and slightly increase the distance between the upper and middle LEDs (pitching down leads to point overlap).

Len62, 2018-06-21:
@sthalik Sounds interesting, is there a thread about it? I found a cure for the pitch issue I was having, I just rotated the clip upwards a slight bit to get a larger separation between the 2 points.

sthalik, 2018-06-23:
@Len62 I didn't publicize it on my own initiative, given two companies sponsoring opentrack received the information. It's not secret, people buy these clips.

You're also able to get higher gain (hence bigger points, more accurate) with points that occlude less. Anyway you should like it.

sthalik, 2018-06-27:
Please check the latest build. There aren't many functional changes, however I rewrote large parts of the mapping code to work on HiDPI without jaggies.

However given the amount of internal changes I'd like you to test your workloads and see if anything's broken.

Builds as always on <<http://opentrack.misaki.pl>>.

sthalik, 2018-06-27:
I added a new feature to the PT tracker that wasn't properly documented. Now all pixels that don't pass the brightness test don't count for center estimation, as if they were all black. It seemed to help on my own workload, but I have no time or will to play sim games. Please drop in if the PT tracker suddenly got more jumpy or anything.

Len62, 2018-06-28:
@sthalik Okay, I tested out the new build, working fine with my set-up. Movement does seem a little more fluid. Still working on the new clip per your previous comment.

cpuwolf, 2018-06-29:
@sthalik I just fetched latest commit, CMake + MSVC 2017 preview + pt/wii work fine. i feel it much more better than before

sthalik, 2018-06-29:
Just to be clear, the change affects only PT. Wii has its own point extractor. Do you have results with PT alone?

MathijsG, 2018-07-17:
Today I spotted your dimensions you're mentioning. Is this what you mean?

[image]

Or did I mix something up?

sthalik, 2018-07-17:
@MathijsG probably... The idea is to basically have a standard clip, but cut depth in half.

sthalik, 2018-07-23:
Updated first post.

Len62, 2018-07-25:
@sthalik No problems with my setup with the 7/16 build. Haven't got around to making that new clip (bathroom renovation almost done).

Len62, 2018-09-18:
@sthalik Just checking to see how you are doing, hope everything is okay.

sthalik, 2018-11-03:
Hey @Len62 and thanks for checking out the latest build. For others, you can also try the yesterday's build since the most important bits have been fixed:

- mapping settings not cancelling when closing dialog
- tracking not working (!)
- some relative translation issues

As usual, http://opentrack.misaki.pl

Len62, 2018-11-04:
@sthalik  Haven't run into any issues with the latest build.

MathijsG, 2018-11-05:
Can we have those testing builds maybe on this website like the official releases, but then from a testing branch or something? I always lose that link :-P. I'll save it now anyway, and will test with my headtracking enabled games.

sthalik, 2018-11-05:
I can't make a separate section for prerelease builds. This link has its own wiki page though.

albertolg, 2018-11-05:
I tested id but seems this problem is still present? https://github.com/opentrack/opentrack/issues/811

sthalik, 2018-11-09:
Today `20181109` marks the day that will live in infamy, the first time anyone took administrative action since the project started in 2012. A user cross-posted to three topics and I removed two of these cross-posts. It doesn't appear to start any trend where issue tracker moderation would be needed though.

Len62, 2018-11-29:
@sthalik 2.3.11 beta 1 is working fine for my set-up. Nothing else to report, will test some more later today.

Len62, 2019-01-14:
@sthalik Beta 2 also working fine.

sthalik, 2019-01-15:
@Len62, cool!

I found the reason why saving in the mapping window sometimes makes everything jump around. It was buried in core code.

MathijsG, 2019-01-15:
Nice, in the weekend I will dive into my headtracking games again and test them with the new beta.

I'm into goold ol' Il-2 Sturmovik 1946 again. When I played that I wasn't aware yet of the headtracking initiatives (mainly freetrack at that time I guess) so I mainly played bombers, which were more easy for me.

By the way, was the Discord channel killed? :-P maybe we should launch something like that, or maybe Slack. I'm willing to give some support to users that way sometimes.

MathijsG, 2019-01-20:
Didn't run into weird issues yet with the tracking!

sthalik, 2019-01-20:
@MathijsG @Len62,
Can you check the latest experimental build? It's to become beta3 or release, but I worry about Accela or even the mappings not working. There's a fix for a rare freeze that's intrusive for all mapping-related stuff.

Len62, 2019-01-20:
@sthalik Just tried out the 20190117 build and did not have any issues.

Len62, 2019-02-04:
@sthalik No problems with 20190203 build on my system.

Len62, 2019-02-28:
@sthalik The Beta 3 is working perfect for my set-up. Using an ini file I had created for the last test build. Everything seems much clearer and well defined in the tracker and octopus windows. Could be a placebo effect but everything seems much smoother. Thanks for all your hard work.

diVineProportion, 2019-05-27:
> To bad no one cares about Linux build :p
> May i ask - i'm trying to switch fully to ubuntu, is there a ppa from you or is there a documentation how to install OpenTrack on Linux?
> Otherwise, i found this one, maybe you can get something useful from it:
> https://github.com/uglyDwarf/linuxtrack

I know that was old, but wine tricks, or whatever allows direct launching of .exe seems to run opentrack fine, my only problem was I don't know enough about hardware or whatever it would take to get my ps3 eye cam to be default. Arch / Antergos AUR repository has opentrack, but even here, the wine emulated one seemed to work better

sthalik, 2019-05-27:
> my only problem was I don't know enough about hardware or whatever it would take to get my ps3 eye cam to be default

Make sure you're using opencv 4.x from the testing repository.

gb2111, 2019-09-06:
what are the plans for this release?

Len62, 2019-10-05:
@sthalik  With the 20190929 build I can't get OT to center. Does not center on startup, and hotkeys have no effect either. Hotkey for start/stop is fine.

sthalik, 2019-10-06:
@Len62, keyboard center works fine for me.

Len62, 2019-10-06:
@sthalik  Oops, chalk that up to operator error, sorry. I do like the new octopus view and everything is running great with my set-up.

sthalik, 2019-10-07:
Glad to hear that. Does the octopus rotate correctly for you (at least with Z disabled)?

Len62, 2019-10-09:
For pitch the movement is fine but, for yaw it appears the axis is located on the left hand side of the octopus instead of the center, same for the Z axis. I haven't tried roll and/or Y axis yet.

MathijsG, 2019-10-27:
Can I help around somewhere? I'm trying out X-Plane 11 nowadays. I got some weird crashes when editing the graphs, but it could be because I used older profiles. Will fiddle with it around more.

sthalik, 2019-10-27:
Send me a minidump please. Use [these instructions](https://support.microsoft.com/en-us/help/931673/how-to-create-a-user-mode-process-dump-file-in-windows).

closed 2020-01-27 by sthalik