opentrack/opentrack #787: FindQt.cmake Error

CJPilot, 2018-05-22:
Hello, I was trying to build opentrack, but I keep getting a FindQt.cmake not found error. 

CMake Error at cmake/opentrack-qt.cmake:1 (find_package):
   By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
   asked CMake to find a package configuration file provided by "Qt5", but
   CMake did not find one.

   Could not find a package configuration file provided by "Qt5" with any of
   the following names:

     Qt5Config.cmake
     qt5-config.cmake

   Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
   to a directory containing one of the above files.  If "Qt5" provides a
   separate development package or SDK, be sure it has been installed.
 Call Stack (most recent call first):
   CMakeLists.txt:44 (include)

CJPilot, 2018-05-22:
Here is the is what I've been following: http://tkoletivrijedi.in.rs/node/219?language=en
I'm on Ubuntu 16.04

sthalik, 2018-05-22:
You need at least

- qttools5-dev
- qttools5-dev-tools
- qtbase5-dev

When building for Wine, it also needs multilib dependencies and `winegcc`.

CJPilot, 2018-05-22:
The dev-tools install fine but qttools5-dev depends on qtbase5-dev, which seems to need a bunch of mesa stuff:
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 qtbase5-dev : Depends: libgl1-mesa-dev but it is not going to be installed or
                        libgl-dev
               Depends: libgles2-mesa-dev but it is not going to be installed or
                        libgles2-dev but it is not installable
               Depends: libglu1-mesa-dev but it is not going to be installed or
                        libglu-dev
E: Unable to correct problems, you have held broken packages.

CJPilot, 2018-05-22:
I'm using this for my GD: https://launchpad.net/~paulo-miguel-dias/+archive/ubuntu/mesa

CJPilot, 2018-05-22:
And I tried sudo apt-get -f install

sthalik, 2018-05-23:
Sorry, I'm not here to provide general Linux help.

closed 2018-05-23 by sthalik