opentrack/opentrack #789: <proc/readproc.h> Missing

CJPilot, 2018-05-23:
Note: I have never built a program as long and as complicated as Opentrack. Sorry for my newbie questions.
When I try to make opentrack, I get an error saying that proc/readproc.h is not found. I have CMake-ed it already without problem. I'm on Ubuntu 18.04, compiling for linux.
What I did so far:
`sudo apt-get install qttools5-dev qttools5-dev-tools qtbase5-dev`
`mkdir build`
`cd build`
`cmake ..`
I pressed c, then g.
`make`
Then I get this error:
`[ 88%] Building CXX object gui/CMakeFiles/opentrack-user-interface.dir/process_detector.cpp.o
In file included from /home/connor/opentrack-unstable/gui/process_detector.cpp:11:0:
/home/connor/opentrack-unstable/compat/process-list.hpp:134:10: fatal error: proc/readproc.h: No such file or directory
 #include <proc/readproc.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
gui/CMakeFiles/opentrack-user-interface.dir/build.make:174: recipe for target 'gui/CMakeFiles/opentrack-user-interface.dir/process_detector.cpp.o' failed
make[2]: *** [gui/CMakeFiles/opentrack-user-interface.dir/process_detector.cpp.o] Error 1
CMakeFiles/Makefile2:1988: recipe for target 'gui/CMakeFiles/opentrack-user-interface.dir/all' failed
make[1]: *** [gui/CMakeFiles/opentrack-user-interface.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2`
Thank you for your time!

CJPilot, 2018-05-23:
Sorry for adding another issue. Turns out you also need to install libprocps-dev.

closed 2018-05-23 by CJPilot

sthalik, 2018-05-24:
No worries! Just next time paste command output like that:

\`\`\`
error
output
\`\`\`

so it appears like

```
error
output
```

sthalik, 2018-05-28:
See #792