opentrack/opentrack #791: Oculus Rift 1.4.0 - erroneous roll while pitching

senorprime, 2018-05-27:
Using opentrack 2.3.10.

Whenever I pitch up or down, I get significant roll, I can't figure out where it's coming from.

For instance at pitch -44, I have a roll of 50, and and about -60 roll when the pitch is 40.

sthalik, 2018-05-27:
Do the other values (yaw and pitch) change when that roll happens?

senorprime, 2018-05-28:
Yes, when I roll the HMD, the pitch significantly increases when rolling left, and decreases when rolling right. Essentially the inverse of what happens when it pitches.

sthalik, 2018-07-06:
What is the raw roll value when you're centered?

senorprime, 2018-07-08:
Very close to 0, it's broadly the same as the game data output, so if I just hold the headset in my hands and carefully roll it, the Raw roll value is pretty much 0, and it's just the pitch that increases/decreases.

sthalik, 2018-07-08:
I'm going to implement a custom center function for Rift. Also Rift earlier to 1.4.0 is going away.

sthalik, 2018-08-03:
What happens if you disable "center at startup" and don't center by yourself? Does it work then?

sthalik, 2018-08-03:
Please head on to #820.

closed 2018-08-03 by sthalik