opentrack/opentrack #792: ubuntu package

sthalik, 2018-05-28:
I'm halfway done with a Ubuntu package for `bionic`. The scripts are badly written, i.e. not Debian-style but they work. I'm mostly missing dependency lists for Debian `sid` and Ubuntu `bionic`.

CJPilot, 2018-05-28:
Does this mean a .deb package?

sthalik, 2018-05-28:
Yes.

CJPilot, 2018-05-28:
Do we have to download the debian folder? Or just sudo apt-get install opentrack?

sthalik, 2018-05-28:
I'm gonna see about making a PPA but first need to finish up the metadata.

sthalik, 2018-05-30:
Pinging @ljgdasfhk

You can try it already if you're adventurous. You're gonna help development by pointing out bugs or just confirming it works.

You need `dpkg-buildpackage`. You can also see that without modifications to `debian/rules`, the package won't pick up Aruco and other modules contained within the `opentrack-depends` repo.

The biggest issue for making it production-ready is the lack of expressed runtime dependencies. If you can build opentrack, you can already run it so it's not a problem.

I made `debian/rules` into dumb shell invocations on purpose. It's easy to modify without knowing Debian's bureaucracy. It's sort of similar to `PKGBUILD`s for Arch or Gentoo's overlay packages. You can add the path to our Aruco or steamvr as simply as editing any other shell script.

I'm gonna add runtime dependency list but this is the proof of concept.

ljgdasfhk, 2018-06-02:
I suggest using the following method of distribution:
[makeself](https://github.com/megastep/makeself)
```
$ makeself.sh --nox11 $HOME/opentrack-build/install $HOME/opentrack.sh "opentrack" ./bin/opentrack
```
then opentrack will be started by the command ```opentrack.sh```

jimiz777, 2018-06-02:
please add Ubuntu xenial package in ppa

closed 2018-08-03 by sthalik