opentrack/opentrack #794: Please add 100 & 50 fps for countries with 50Hz electric signal.

calico2, 2018-06-04:
In my country we have 50 Hz signal and when artificial illumination is in use, it makes an awfull blink effect when captured at 30 60  75 or 120 fps. Ps3 eye toy driver can handle 50 or 100 fps smothly but the option list has not got these options. I am using aruco tracker.  1000 thanks in advance.

sthalik, 2018-06-04:
Changing fps won't help. Only driver-side power line compensation will help. These are different things.

I'd like to see a video with the power line flicker visible and run some tests. Please upload one, just compress it first.

calico2, 2018-06-04:
[prueba60.zip](https://github.com/opentrack/opentrack/files/2068843/prueba60.zip)

calico2, 2018-06-04:
I am trying to make it lower than 10 MB  rename this one as z01

[prueba60.z01.zip](https://github.com/opentrack/opentrack/files/2068855/prueba60.z01.zip)

calico2, 2018-06-04:
rename this as z02 
[prueba60.z02.zip](https://github.com/opentrack/opentrack/files/2068859/prueba60.z02.zip)

calico2, 2018-06-04:
If i use point tracker 1.1  I can set capture to 50 Hz  and adjust in 1 Hz steps. The capture is perfect then. In aruco I am limited to fixed FPS.

sthalik, 2018-06-05:
I'm unable to open it using 7zip or unzip from Unix. Please send something that extracts using either of these tools.

I recommend just splitting the data in half or hosting it on Dropbox.

sthalik, 2018-06-05:
Do you get a bad capture in PointTracker if Hz isn't a multiple of the European 230V/50Hz AC rate?

calico2, 2018-06-05:
[prueba60.7z.001.log](https://github.com/opentrack/opentrack/files/2071919/prueba60.7z.001.log)
[prueba60.7z.002.log](https://github.com/opentrack/opentrack/files/2071924/prueba60.7z.002.log)
[prueba60.7z.003.log](https://github.com/opentrack/opentrack/files/2071929/prueba60.7z.003.log)
[prueba60.7z.004.log](https://github.com/opentrack/opentrack/files/2071938/prueba60.7z.004.log)

calico2, 2018-06-05:
Please rename the files without the .log for extracting.

commit f95a4641 (2018-12-07): tracker/aruco: add more camera fps options

These coincide with power line frequencies.

Fixes #794

Users will need to adjust existing settings.
closed 2018-12-07 by sthalik