opentrack/opentrack #796: [Feature] Implement donation system on this website

MathijsG, 2018-06-08:
Hey Sthalik,

I'm a little saddened by the few responses to being a help to this projects, which is widely used. We've talked earlier about donations, isn't it a possibility to add it to this website already? That would be just a matter of adding information or text about, and maybe include a payment system. Many many other projects on Github and other websites have this as well.

Although I'm in lack of time currently :( I would like to contribute this way, and I'm guessing I'm not the only one.

I don't know if you like Paypal, but they have an easy button you can implement:
https://developer.paypal.com/docs/classic/paypal-payments-standard/integration-guide/donate_step_1/

sthalik, 2018-06-09:
Hey,

Thanks for caring about the project.

What the project *needs* the most is someone else to step in and work on the core code. Having it dependent upon one person is what's keeping updates slow. I'm burnt out on this codebase due to having the sole responsibility.

For the duration it's pointless for me to receive any donations as it won't make things get done any faster. Having the sole responsibility over such an extended period of time is burning me out. Some things should've been done a year ago, like Tobii support.

MathijsG, 2018-06-09:
The donation is more to contribute back to the developments you did for years. Even if you'd stop tomorrow I still would donate. It's not a either, or situation, so I hope you will consider to support a certain way of donating, at least on the website (easiest to implement).

Len62, 2018-06-10:
While I agree whole heartedly with MathijsG's idea, I think once people donate, their expectations are for more frequent updates/improvements. But like MathijsG I would also donate to support past, and future, developments.

sthalik, 2018-06-10:
Both of your contributions through bug reports, assisting users, and feature requests are - comparatively - worth several hundred bucks in donations.

I agree about the expectations bit. The issue tracker has been calm so far but users' feelings of entitlement may possibly require me to take administative action. Funny, over the years not once had I deleted anything or locked an issue.

sthalik, 2018-08-04:
Thank you for the feedback. I won't be asking for donations in any form.

closed 2018-08-04 by sthalik

MABlosfeld, 2021-11-15:
Creating a form of donation is not asking for a donation, it is just providing the possibility for all those who want to help. I would like to step in and work on the code but I am just a retired DELPHI programmer. Projects like FREEPIE/OPENTRACK/MOONLIGHT/DEPTH3D are all excellent projects and deserve our full support.