opentrack/opentrack #798: TrackIR and constant IR source exposure

cloutech, 2018-06-11:
Has anyone opened up their TrackIR and took a current reading on the IRLEDs? As I understand they use IRLEDs and point it straight at you, your reflectors will provide the source for tracking. If they use similar IRLEDs I use (osram or temic, industrial grade, high power, pretty cheap, 950nm) than at that distant you would be staring into a IR source the entire time while playing, meaning your retina is exposed. Depending on the power this could damage your retina pretty seriously over time resulting in burned retina tissue, because your eye sensory system cannot detect wavelength of that magnitude and thus your safety features (eye lid and iris) wont work. 

So, anyone took a reading?

cloutech, 2018-06-11:
I set my curcuit to less than 80% current and I am pulsing the IRLEDs. Ofc we can assume the maximum without pulsing, which will damage your retina for sure. I dont think Osram and all other manufacturers just put that warning in the datasheet for fun. Yes they are over-carefule with stuff like that, but there is a risk here. 
Osram - SFH4346:
"Notes 
Depending on the mode of operation, these devices
emit highly concentrated non visible infrared light
which can be hazardous to the human eye. Products
which incorporate these devices have to follow the
safety precautions given in IEC 60825-1 and IEC
62471."

sthalik, 2018-06-11:
Compare to looking directly into a visible light LED with same eV photon energy.

How much does your whole LED array take in watts?

cloutech, 2018-06-11:
I followed the example in the doc I linked and it comes out ok, as long as you keep a distance of more than 10cm. Even exposure for upto 2h was ok for 3 IR LEDs. Maybe I was just over-cautious.

sthalik, 2018-06-11:
Two hours isn't enough. I'd expect not being aware that the emitter is still enabled. What's the actual value you got? Also the typical distance from display to the user is 50-80 cm. 

It's strange, 80mW is barely any energy, especially IR. Similarly cell phone towers or radio don't make me go blind.

cloutech, 2018-06-11:
80mW/m^2 being Ee, corneal exposure, which not only takes time and distance into account, but also the entire area of the corneal (which is pretty big and seeing how it would also be cooled by occasional blinking). 

If we just do a quick and dirty calc the 3 IR LEDs should be about 360mW (=1.6V * 0.075A * 3) - not pulsed, at 75% of the forward current I_f. I mean they do allow 1A for very short pulses (usec), but I do not think anyone would need that, unless you want your signals to rival sunlight.

sthalik, 2018-06-12:
This article looks sensible: <<https://www.tacticalinfosys.com/blog/2017/9/11/ir-illumination-eye-safety>>. See summary at the very bottom of the article.

An IR LED is omnidirectional thus different than a focused beam. Interestingly, one can find eye-safe rangefinders. They use 10Hz focused beams.

Your results with mW being the limit even assuming perfect efficiency (and a non-focused beam), don't add up. Emitting fraction of a Watt doesn't make sense by looking at orders of magnitude, regardless of the spectrum you're emitting in (unless you ionize, heh).

My IR clip has somewhat focused LEDs with USB input wattage. I think they take 1.5W after resistors. They don't even get hot. I'd expect your emitters needing to get hot at all to be reflected at relevant S/N ratio. As well as taking few orders of magnitude more power, all while still having even more orders of magnitude safety for 25cm constant exposure.

Also you *do* need to exceed sun's IR that bounces from walls to detect the reflectors.

cloutech, 2018-06-12:
It does add up pretty well. I also use USB at about 5.004 V and my resitors get hot. That is where I lose most power, but like I said, I reduced the emission to around 70% which lowers wavelength to about 920-930nm. And yes, as a result I cannot use it without closing the shades to limit sun IR. But that was my intention: it is safe for me to wear, although since the emission half angle is only 20 wearing it was never the issue. But it is also safe if it gets reflected by a shiny surface or I look at someone who is standing in front of me about 1m away. 
And I would also caution to not use any LED at maximum (continous) forward current as it will burn out your LED more quickly. Reduce it at least 20% and that LED will hold a lifetime.

One mistake I did make was the 80mW/m^2, it is in fact something around (corneal) E_e=0,1W/m^2 < 100W/m^2 for one IR LED and (burn hazard) Lr = 41.009,0 W/m^2/rs < 5.427.963 W/m^2/rs and (retinal) L_ir = 41.008 W/m^2/rs  < 6*E6 . All at a distance of 1000mm and 7200s = 2h exposure.

sthalik, 2018-06-12:
Consider plugging in visible light LEDs operating at same power. You'd know directly if it causes any discomfort. The pain will tell you 😄. For instance the stock PS3 Eye red light emitter is painful to look at from 60-80 cm when mounted on top of the display.

I recommend not going further into infrared. The CMOS sensors won't detect that as well as when closer to the visible spectrum. Companies I know use 850 nm and that works well with the PS3 Eye. Given your goal of maximizing the s/n ratio while keeping within safety margins, you'd want max detection gain for the sensor.

I wonder what reflective surface is the one you chose. Can you share the product name or material properties?

Interestingly, one of the companies that sent hardware for opentrack keeps a single clip running continuously for years. After 5 years it's still working with no faults. I haven't imagined that despite their input power margin of some 5-15%.

sthalik, 2018-06-12:
> Doesn't this perceived problem only exist with the original TrackIR? Where the camera is an actual IR transmitter and the cap only is a passive refector?

@cloutech specifically wants to make a passive reflector with the correct safety margin.

sthalik, 2018-06-12:
[Sorry for triple-post. You'd not get the edited content via notifications.]

@cloutech can you write out the intermediate terms in your equations please? It's a bit hard to follow. Also which formulae are you using?

@cloutech also does max exposure time scale linearly with input power? It's confusing because convection etc.

cloutech, 2018-06-12:
The equations are in the .pdf I linked. They also offer a calculator at the bottom of the doc (link to an excel sheet) which you can use to get a hang of things. Just somehwat adjust the circuit they offer to fit your power levels. 
And yes, the problem only exists with the TrackIR or Wii-Sensorbar or when you have an IR clip and your daughter comes running into the room and you look at her, bam, you point that thing right at her at a very short distance. Not funny. And of course all sorts of reflections from shiny areas (e.g. screen cases shiny like piano finish, anyone?). 
Btw. I read that the wiimote sensor operates at the 950nm range which was my first plan, but since the drivers for the wiimote are really bad I switched to a simple cam and put a filter on it.

sthalik, 2018-06-12:
Is our Wii tracker also bad? I have no experience with the hardware but it's supposedly good enough. I'd be happy to hear opinions.

cloutech, 2018-06-12:
no, the wii tracker is great, its just that the drivers can be a big problem (bluetooth stack). PS3 is also problamatic for the same reason. 

Dont worry, you are not the smart guy. There are plenty of examples of things being sold that later turned out to cause harm one way or another. And there are two ways to find out about it: having a lot of people report eye sight reduction or even blindness or some guy somewhere doing the math again and re-evaluating it. What's there to lose if going thru the math again? It will give insight for future designs.

cadahl, 2018-08-31:
Since IR is in the range of heat radiation emitted by the sun, is the little IR diode even comparable to spending a day on the beach?

Someone enlighten me please. I would have assumed the energy from such diodes to be negligible in comparison.

cloutech, 2018-09-01:
The sun is dangerous because of UV rays, not so much IR, hence you do not really have a problem behind a window that filters the UV. And we were talking about the case where the source of IR is somewhat directly pointing at you (above your screen) and thus the rays would also be able to enter your eye directly thru your lense and hit your retina. This will heat up your retina, but the conclusion was that the tissue is able to dissipate the energy quickly enough so no harm should be caused. 
Just do the math and follow the linked article. There are ways to misuse IR diodes that can cause harm. When I buy IR diodes they come with a warning label that they can cause eye damage. You dont need huge amounts of energy if the area of affect is that small and even gives you a lense to focus beams ;)

sthalik, 2018-09-01:
> When I buy IR diodes they come with a warning label that they can cause eye damage

With lenses, not omnidirectional emitters. We're talking about 2.5A 5V for the array. Unless you plan to connect it elsewhere than the USB port.

cadahl, 2018-09-01:
Also the Wii sensor bar is a bunch of IR LEDs pointing at you. Maybe there's a difference in power as it doesn't rely on bouncing the light off of your face though.

closed 2019-02-11 by sthalik