opentrack/opentrack #800: easy DIY reflective clip idea

sthalik, 2018-06-12:
Hey,

First off see PS3 Eye vs mirror with sun obscured by clouds:

[image]

Now consider an incandescent lightbulb EM spectrum. Image credit: <<https://www.comsol.com/blogs/calculating-the-emission-spectra-from-common-light-sources/>>.

[image]

Then consider typical cloth-like lamp covering to reduce total light emitted. Staring at a covered 60W lightbulb doesn't cause any strain or afterimage.

Finally consider a clip made of small reflective material at correct angles :)

Benefits:
- no wires
- easy construction w/ no electrical knowledge or even soldering
- easy notification it's on, to flip it off
- only parts needed are from disassembling a lamp or a wall light
- sun doesn't make it worse, it makes the light brighter
- more efficient light source possible, depending on emitted EM spectrum
- works without an IR filter
- low webcam exposure needed
- very small input wattage needed for enough brightness (but still 10+ Watts)

Disclaimer: I got this idea while reading #798

Pinging: @Len62 @MathijsG

sthalik, 2018-06-12:
Also pinging @mrsanchos

bjensie, 2018-06-12:
@sthalik I've earlier had the idea that after printing the aruco-pattern, use a reflective tape and mount it to the white portions of the pattern, including the part around the actual pattern, Shouldn't that enhance the contrast when having a pretty heavy change in lighting environment, both conventional lamps and also sunlight? So far I haven't tested the idea, would be nice to get an advice from you before that!

sthalik, 2018-06-13:
@bjensie I'm thinking of using a matte material for Aruco. With too high exposure it's all white otherwise, assuming paper and an ink printer.

closed 2018-08-03 by sthalik