opentrack/opentrack #801: Model position gets displaced after loosing 1 point momentarily

nuno-pc, 2018-06-12:
Using: hat model; input pointracker 1.1; output freetrack 2.0 enhanced

If turning left/right it looses track of one point on left/right:
Then (when getting back to center and reacquiring point) all is mixed up; view is stuck,
and Model position is out of place.
Need to press re-center twice to get back to normal.

Attached pictures:

[IssueOT_Losing_Point_Displaces_Model_Center_20180612.pdf](https://github.com/opentrack/opentrack/files/2094785/IssueOT_Losing_Point_Displaces_Model_Center_20180612.pdf)

Len62, 2018-06-12:
I would try increasing the minimum point size to 6 or 7, and max to 9 or so. With small point sizes some tracking is lost if it gets to low. Experiment with the point settings.

sthalik, 2018-06-13:
- Look forward and center after this happens in worst-case scenario.
- Increase dynamic pose timeout.

sthalik, 2018-06-13:
> Need to press re-center twice to get back to normal.

If you need to center **twice**, it's a bug.

sthalik, 2018-08-04:
Pressing "center" once is now enough in the dev build.

closed 2018-08-04 by sthalik