opentrack/opentrack #802: mapping or/and accela problems

PAL123, 2018-06-19:
hello,

I tried to use opentrack, but there is something that bothers me a lot. when i try to transfer my settings from FaceTrackNoIR 1.7 with Pointtracker 1.1 to opentrack. then I notice that something to the curven or synonymous with accela is not working properly.

Here is an example of my FaceTrackNoIR settings, I use a 150 degree rotation output to 20 degrees rotation input. and I have a slight initial turn to gently look around the cockpit. but I do not succeed in opentrack with the best of intentions. opentrack moves very jerkily. and you also have a deadzone even if you remove it in accela ..., it would also be very nice if you could display the curve mapping legibly in the .ini file so that you could better type in it by hand.
you can try the comparison with FaceTrackNoIR 1.7 and try to transfer this attitude to opentrack:

Below are 2 examples of my problem

example setup FaceTrackNoIR 1.7 for il-2 cliffs of Dover:

[Tracking]
Smooth=1
invertYaw=true
invertPitch=true
invertRoll=false
invertX=false
invertY=false
invertZ=false

[GameProtocol]
Selection=2
DLL=FTNoIR_Protocol_FT.dll

[TrackerSource]
Selection=1
DLL=FTNoIR_Tracker_PT.dll
2ndDLL=None

[Filter]
DLL=FTNoIR_Filter_Accela.dll

[PointTracker]
CameraId=0
CameraF=@Variant(\0\0\0\x87?\x80\0\0)
CameraResX=640
CameraResY=480
CameraFPS=75
CameraRoll=0
CameraPitch=0
CameraYaw=0
PointExtractThreshold=137
PointExtractMinSize=2
PointExtractMaxSize=30
PointModelM01x=@Variant(\0\0\0\x87\0\0\0\0)
PointModelM01y=@Variant(\0\0\0\x87\x42 \0\0)
PointModelM01z=@Variant(\0\0\0\x87\xc1\xf0\0\0)
PointModelM02x=@Variant(\0\0\0\x87\0\0\0\0)
PointModelM02y=@Variant(\0\0\0\x87\xc2\x8c\0\0)
PointModelM02z=@Variant(\0\0\0\x87\xc2\xa0\0\0)
tMHx=@Variant(\0\0\0\x87\0\0\0\0)
tMHy=@Variant(\0\0\0\x87\0\0\0\0)
tMHz=@Variant(\0\0\0\x87\0\0\0\0)
DynamicPoseResolution=true
VideoWidget=true
SleepTime=10
ResetTime=1000
EnableRoll=true
EnablePitch=true
EnableYaw=true
EnableX=true
EnableY=true
EnableZ=true

[PointTrackerDialog]
ActiveModelPanel=0
CustomM01x=0
CustomM01y=40
CustomM01z=-30
CustomM02x=0
CustomM02y=-70
CustomM02z=-80
ClipTopHeight=40
ClipTopLength=30
ClipBottomHeight=70
ClipBottomLength=80
CapHalfWidth=40
CapHeight=60
CapLength=100

[Accela]
Reduction=20

[Curves-Accela-Scaling-Rotation]
point-count=4
point-0-x=0.1
point-0-y=0
point-1-x=1.43
point-1-y=2.45
point-2-x=2
point-2-y=5.44
point-3-x=2.06
point-3-y=6

[Curves-Accela-Scaling-Translation]
point-count=10
point-0-x=0
point-0-y=0
point-1-x=0.282722513089005
point-1-y=0.08
point-2-x=0.492146596858639
point-2-y=0.306666666666667
point-3-x=0.764397905759162
point-3-y=0.84
point-4-x=1.00523560209424
point-4-y=1.62666666666667
point-5-x=1.17277486910995
point-5-y=2.78666666666667
point-6-x=1.25130890052356
point-6-y=3.6
point-7-x=1.31937172774869
point-7-y=4.29333333333333
point-8-x=1.38219895287958
point-8-y=4.90666666666667
point-9-x=1.43455497382199
point-9-y=5.65333333333333

[FT]
UsedInterface=0

[FTIR]
useTIRViews=false
useDummyExe=true

[Curves-Yaw]
point-count=5
point-0-x=1
point-0-y=3.75
point-1-x=2
point-1-y=7.5
point-2-x=3
point-2-y=11.25
point-3-x=6
point-3-y=22.5
point-4-x=20
point-4-y=150

[Curves-PitchUp]
point-count=5
point-0-x=1
point-0-y=3
point-1-x=2
point-1-y=6
point-2-x=3
point-2-y=9
point-3-x=6
point-3-y=18
point-4-x=15
point-4-y=90

[Curves-PitchDown]
point-count=5
point-0-x=1
point-0-y=3
point-1-x=2
point-1-y=6
point-2-x=3
point-2-y=9
point-3-x=6
point-3-y=18
point-4-x=15
point-4-y=90

[Curves-Roll]
point-count=1
point-0-x=15
point-0-y=60

[Curves-X]
point-count=1
point-0-x=50
point-0-y=10

[Curves-Y]
point-count=1
point-0-x=50
point-0-y=30

[Curves-Z]
point-count=1
point-0-x=50
point-0-y=100

---------------------------------------------------------

almost same Setup for opentrack:

[migrations]
last-migration-at=20180118_00~

[opentrack-ui]
keycode-center=Home

[spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2@$\0\0\0\0\0\0@9\0\0\0\0\0\0@4\0\0\0\0\0\0@b\xc0\0\0\0\0\0)

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2@$\0\0\0\0\0\0@4\0\0\0\0\0\0@.\0\0\0\0\0\0@V\x80\0\0\0\0\0)

[tracker-pt]
camera-fps=75
camera-name=PS3Eye Camera
model-centroid-x=0
model-centroid-y=0
model-centroid-z=0

[accela-sliders]
rotation-deadzone=@Invalid()
rotation-sensitivity=@Invalid()
ewma=@Invalid()
rotation-nonlinearity=@Invalid()

[modules]
filter-dll=Accela

[opentrack-mappings]
roll-max-value=15
x-max-value=100
y-max-value=100
z-max-value=100

[spline-X]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@I\0\0\0\0\0\0@$\0\0\0\0\0\0)

[spline-Y]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@I\0\0\0\0\0\0@>\0\0\0\0\0\0)

[spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@I@\0\0\0\0\0@R\x80\0\0\0\0\0)

sthalik, 2018-06-20:
May be few things. The default Accela settings are sluggish, if anything.

- use one of output protocols, not both with that game
- adjust PT settings like threshold, etc.
- see if Windows 10 has Game Mode enabled, see if disabling helps (someone reported that)
- check with different game

Adjusting the gain manually won't be available. Barely anyone understood it, and even fewer knew the significance of the gain spline.

sthalik, 2018-06-20:
You don't have any point model settings in opentrack. In fact PT runs at default clip siz.e

That's how the rot gain looks like now. I expect there's something else wrong here.

[image]

The left half of the graph is for the minor movement. You can see that some parts are basically flat, but non-zero. This is on purpose.

Do you have problems with *all* Accela slider positions, or only from a certain point?

Keep the deadzone setting at `.06`, don't go over `.1` unless you get it working first.

PAL123, 2018-06-20:
I have problems with all accela settings. although I have not set a deadzone is still present. and no matter how I set a curve in the mapping, it smears or it's jerky. my goal is to make a fine initial curve from the middle into every rotation. then I do not need a dead zone, but it is still there if only minimal. I also tried 60hz as recommended, but this is not the problem

how do you get the accela rotation gain in the opentrack displayed as in the picture?

example:
In  FaceTrackNoIR this works nice, but absolutly not in opentrack:
[Curves-Yaw]
point-count=3
point-0-x=0
point-0-y=0
point-1-x=6
point-1-y=22.5
point-2-x=20
point-2-y=150

[Curves-PitchUp]
point-count=3
point-0-x=0
point-0-y=0
point-1-x=6
point-1-y=22.5
point-2-x=15
point-2-y=90

[Curves-PitchDown]
point-count=3
point-0-x=0
point-0-y=0
point-1-x=6
point-1-y=22.5
point-2-x=15
point-2-y=90

PAL123, 2018-06-20:
maybe the problematic is that on the one hand the mapping grid is too coarse. and the other thing is that you use another accela, which is not the same as in FaceTrackNoIR, where you seem to be able to learn more about the .ini.

my settings are in FaceTrackNoIR for accela this, but how do i get eg. this in the accela of opentrack?

[Accela]
Reduction=20

[Curves-Accela-Scaling-Rotation]
point-count=4
point-0-x=0.1
point-0-y=0
point-1-x=1.43
point-1-y=2.45
point-2-x=2
point-2-y=5.44
point-3-x=2.06
point-3-y=6

[Curves-Accela-Scaling-Translation]
point-count=10
point-0-x=0
point-0-y=0
point-1-x=0.282722513089005
point-1-y=0.08
point-2-x=0.492146596858639
point-2-y=0.306666666666667
point-3-x=0.764397905759162
point-3-y=0.84
point-4-x=1.00523560209424
point-4-y=1.62666666666667
point-5-x=1.17277486910995
point-5-y=2.78666666666667
point-6-x=1.25130890052356
point-6-y=3.6
point-7-x=1.31937172774869
point-7-y=4.29333333333333
point-8-x=1.38219895287958
point-8-y=4.90666666666667
point-9-x=1.43455497382199
point-9-y=5.65333333333333

sthalik, 2018-06-20:
Again I don't see any clip settings in the opentrack config you posted. This is the important bit.

PAL123, 2018-06-20:
sorry i dont know what you exactly Need

this is my full config for opentrack:
[migrations]
last-migration-at=20180118_00~

[opentrack-ui]
keycode-center=F1

[spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x3\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\x18\0\0\0\0\0\0@6\0\0\0\0\0\0@4\0\0\0\0\0\0@b\xc0\0\0\0\0\0)

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x3\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\x18\0\0\0\0\0\0@6\0\0\0\0\0\0@.\0\0\0\0\0\0@V\x80\0\0\0\0\0)

[tracker-pt]
camera-fps=60
camera-name=PS3Eye Camera
model-centroid-x=0
model-centroid-y=0
model-centroid-z=0

[accela-sliders]
rotation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0?\xc9\x99\x99\x99\x99\x99\x9a)
rotation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0?\xf8\0\0\0\0\0\x1?\xa9\x99\x99\x99\x99\x99\x9a@\x4\0\0\0\0\0\0)
ewma=@Invalid()
rotation-nonlinearity=@Invalid()

[modules]
filter-dll=Accela

[opentrack-mappings]
roll-max-value=15
x-max-value=100
y-max-value=100
z-max-value=100

[spline-X]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@I\0\0\0\0\0\0@$\0\0\0\0\0\0)

[spline-Y]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@I\0\0\0\0\0\0@>\0\0\0\0\0\0)

[spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@I@\0\0\0\0\0@R\x80\0\0\0\0\0)

[spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x3\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\x18\0\0\0\0\0\0@.\0\0\0\0\0\0@.\0\0\0\0\0\0@N\0\0\0\0\0\0)

sthalik, 2018-06-20:
Delan has all the settings same as opentrack's default?

PAL123, 2018-06-20:
[image]
 same settings

sthalik, 2018-06-20:
So what are your settings? Mine are: 

[image]

[image]

Sorry for enlarged images, this is hidpi.

If your points are smaller than 10-13 px, you may want to go for "rotation: .45" in Accela, and put deadzone to `.1`.

Also once you switch to 60Hz the points are going to be bigger and more evenly bright. See:

[image]

PAL123, 2018-06-20:
[image]

[image]

[image]

and that is recommended to accela:
http://ftnoirpt.sourceforge.net/

[Accela]
Reduction=20

[Curves-Accela-Scaling-Rotation]
point-count=4
point-0-x=0.1
point-0-y=0
point-1-x=1.43
point-1-y=2.45
point-2-x=2.0
point-2-y=5.44
point-3-x=2.06
point-3-y=6

PAL123, 2018-06-20:
have tried your settings, but it comes to the same result. should I put my whole FTNoIR pack as a zip so that you can compare it with opentrack?
you will be very convinced of that and then determined the difference or the problem better recognize

PAL123, 2018-06-20:
https://ufile.io/g5cr9

PAL123, 2018-06-20:
the only feature i miss in ftnoir is the relative Translation like rotating +180 Degrees yaw and moving backwards results in moving Forward as a result of that rotation

sthalik, 2018-06-20:
I'm still not completely sure what your problem is. You haven't described it precisely. I need

1. A screenshot of the main interface while it's tracking
2. A gameplay video with the issue visible.

PAL123, 2018-06-20:
between 8-9px

sthalik, 2018-06-20:
Increase gain to about half. You should get 10-12 pixel radius. Can you confirm the new radius?

Len62, 2018-06-20:
Have you calibrated your set-up? Here is the PT section from my config:
[tracker-pt]
camera-fps=100
camera-name=PS3Eye Camera
dynamic-pose-resolution=false
max-point-size=15
min-point-size=5
model-centroid-x=-91
model-centroid-y=-90
model-centroid-z=56
clip-by=66
clip-bz=80
camera-fov=62
clip-ty=43

I noticed your config file has items missing in this section, namely the model-centroid and clip data.

PAL123, 2018-06-20:
https://youtu.be/YwwZp7RFtns

this is Il 2 Sturmovik Cliffs Of Dover Blitz - FaceTrackNoIR, There is a very little deadzone visible, but it does not bother

https://youtu.be/dYPtGgAUEcw

and this is Il 2 Sturmovik Cliffs Of Dover Blitz - OpenTrack, there is a bigger deadzone but in accela i turned off, and the acceleration is a bit stronger than FaceTrackNoIR, but with smoothing, for example. 2.5 degrees I could reduce the. but a higher value than 2.5 degrees could improve it

PAL123, 2018-06-20:
Thank you for the good sake to help me but I also already calibrated the calibrating

PAL123, 2018-06-20:
could you create an opentrack test version with the same accela as in FaceTrackNoIR?
maybe then the enormous deadzone disappears ...

and a higher smoothing value than 2.5 degrees eg max 5 degrees?

PAL123, 2018-06-20:
this is how accela Looks with the recommended Setting from PointTracker 1.1

[image]

sthalik, 2018-06-21:
@PAL123,

There's something horribly wrong here, like having the entire tracking range chopped up into pieces. Even a very high deadzone setting behaves differently in Accela! This doesn't look like a filter problem but as if framerate was cut in a third or something. Or as if the yaw mapping was a staircase. Don't get me wrong, this is clearly aberrant and I'll try my best to help you.

Since you consider this a filter problem, try Kalman at default settings. Not as a permanent solution but to eliminate one of the possibilities (filter or non-filter problem).

Also please note that users are generally satisfied with the Accela filter. I get very few complaints on the issue tracker and the Russian IL-2 forum. The current gain value has been there for a while to keep jumpiness or staircase-like movement to a minimum. 

> could you create an opentrack test version with the same accela as in FaceTrackNoIR?

No. That's pointless given how silly the old Accela was. I wrote that around 2013 having little idea what I was doing. Fortunately you haven't seen the mk1 version.

> and a higher smoothing value than 2.5 degrees eg max 5 degrees?

What smoothing value is the recording based on? Unless the value is something like `0.05` of a degree it's wrong anyway. Just curious. At higher smoothing values you'll get absurd "slow then jump to head position" behavior. This is the same in Accela mk2 as if you made very flat gain with a low slope up to 3, then suddenly jumped up at 3.5 till 4. With no "reduction" at that.

@Len62,
> I noticed your config file has items missing in this section, namely the model-centroid and clip data.

He says delanclip has exact default clip size.

PAL123, 2018-06-21:
the above video from opentrack has the standard smoothing value of 1.5, although I like 2.5 better, but the value is a little too low for my opinion.

Here is a video with the suggestion smoothing 0.05 which makes the whole thing even worse for me, because the movements are too fast.

To rule out misunderstandings, I mention again, it bothers me that you can not really disable **+++ the deadzone and the movements are too hectic compared to FaceTrackNoIR +++**. and you do not get the hectic synonymous over a curve with the same grades as eg from as in FaceTrackNoIR away. It is nice if you can watch the cockpit with a slow movement without blurring it (eg curve with 22.5 degrees left and right = 45 degrees slow movement), but still you can look around 150 degrees left and right or 90 degrees ( same curve as above) up and down
[image]
[image]

in the 2 videos above, I showed how nice slow the movement is on the cockpit directed in FaceTrackNoIR for me and I do not get this transferred in Opentrack

PAL123, 2018-06-21:
hier ein video mit smoothing 2.5 grad, es ist nach meiner meinung viel angenehmer aber die totzone ist wie immer vorhanden in der mitte. wäre der wert ein wenig höher wie 2.5 könnte man es sehr gut für sich selbst einstellen:

https://youtu.be/3icTXRfvY6Y

sthalik, 2018-06-21:
Try Kalman first please.

PAL123, 2018-06-21:
already tried, but what can I say. also find there no right setting

PAL123, 2018-06-21:
Kalman: with 0.600 Degree Rotation it feels not bad, but the deadzone is still present

PAL123, 2018-06-21:
here a Video from the deadzone with Kalman on 0.600 Degree Rotation:

https://youtu.be/ucTMAJnIYHQ

PAL123, 2018-06-21:
ok, I found out something. if I use the desired value of 6x22 in the curve mapping, as I have the value in FaceTrackNoIR. then a dead zone is created. my idea of this value corresponded to a mathematical calculation. to halve the original value at point 6. so that the curve mapping in opentrack does not seem to work properly.

But if I set the value 32 at point 6 suddenly there is no dead zone and the movement is very smooth in the middle

PAL123, 2018-06-21:
here a the Video of: 0x0 - 6x32 - 20x150
https://youtu.be/LQXGFlZPtmc

sthalik, 2018-06-21:
Please send me the .ini `[section]` with the mapping data, when it's broken. Or paste the whole config, like this:

\`\`\`
foo
\`\`\`

will turn into

```
foo
```

then it won't lose formatting.

PAL123, 2018-06-21:
**broken because 6x22 in yaw curve, when you move the Point to 6x32 ist fine:**

[migrations]
last-migration-at=20180118_00~

[opentrack-ui]
keycode-center=F1
neck-enable=false
relative-translation-mode=0

[spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x3\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\x18\0\0\0\0\0\0@6\0\0\0\0\0\0@4\0\0\0\0\0\0@b\xc0\0\0\0\0\0)

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@.\0\0\0\0\0\0@V\x80\0\0\0\0\0)

[tracker-pt]
camera-fps=60
camera-name=PS3Eye Camera
model-centroid-x=0
model-centroid-y=0
model-centroid-z=0
max-point-size=50
min-point-size=2.5
threshold-slider=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0@`\0\0\0\0\0\0\0\0\0\0\0\0\0\0@o\xe0\0\0\0\0\0)
automatic-threshold=true
camera-res-height=480
camera-res-width=640

[accela-sliders]
rotation-deadzone=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0?\xc9\x99\x99\x99\x99\x99\x9a)
rotation-sensitivity=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0@\x4\0\0\0\0\0\0?\xa9\x99\x99\x99\x99\x99\x9a@\x4\0\0\0\0\0\0)
ewma=@Invalid()
rotation-nonlinearity=@Invalid()

[modules]
filter-dll=Accela

[opentrack-mappings]
roll-max-value=15
x-max-value=100
y-max-value=100
z-max-value=100

[spline-X]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@I\0\0\0\0\0\0@$\0\0\0\0\0\0)

[spline-Y]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@I\0\0\0\0\0\0@>\0\0\0\0\0\0)

[spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@I@\0\0\0\0\0@R\xc0\0\0\0\0\0)

[spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x3\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0@\x18\0\0\0\0\0\0@.\0\0\0\0\0\0@.\0\0\0\0\0\0@N\0\0\0\0\0\0)

[kalman-filter]
noise-rotation-slider=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0?\xe4z\xe1G\xae\x14{\0\0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0)
noise-position-slider=@Variant(\0\0\0\x7f\0\0\0\x18::options::slider_value\0?\xe0\x14z\xe1G\xae\x14\0\0\0\0\0\0\0\0?\xf0\0\0\0\0\0\0)

sthalik, 2018-06-21:
I took the liberty of formatting the config data in your post. Now have to sit and think till it's fixed.

Please help me narrow down the issue in the spline code. Add a point between 10 and 20 when the line is almost straight. The idea's to have at least 4 points (`0, 0` is invisible). The numerical instability needs fixing either way.

PAL123, 2018-06-21:
you should have same Problems as me, the only Thing i know is, if the Y Point smaller than 30 a deadzone is noticeable

sthalik, 2018-06-21:
Sorry wasn't the problem on the yaw axis?

PAL123, 2018-06-21:
it seems on all axis when when you ask me, when all are same calculated. for pitch im also sure there is the same issue

I only wanted eg. in the middle of the yaw rotation have a 90 degree angle (45 left + 45 right) in which the movement is halved to the normal 150 degrees output to 20 degrees input per side.

so my mathematic was for each side x20: y150 = 0.13333333x45 = x6, at x6 you have y45: 2 = y22.5 so this is why i came for the yaw axix on curve x6 * y22.5

PAL123, 2018-06-21:
Maybe the problems come from the rough grid

PAL123, 2018-06-21:
in FaceTrackNoIR you can set many numbers behind the comma

OpenTrack has only half value behind comma x , and only full number for y

sthalik, 2018-06-21:
@PAL123,

1. What is your CPU?
2. How many FPS do you have in that game?

I copied your config except PT settings and it all works without that jerky movement. I can't reproduce it. It rotates smoothly without problems.

> so my mathematic was for each side x20: y150 = 0.13333333x45 = x6, at x6 you have y45: 2 = y22.5 so this is why i came for the yaw axix on curve x6 * y22.5

Sorry, I don't get it. How is this significant?

> in FaceTrackNoIR you can set many numbers behind the comma for each x y axis
> OpenTrack has only half value behind comma x , and only full number for y

This is only to make the main UI more readable. Similarly, opentrack shows the video feed at 12 Hz while still doing head tracking up to 250 Hz.

> this is why i asked for readable and editable .ini for these settings

Maybe, but don't count on it. It makes code more simple. Simpler the code, less the bugs.

PAL123, 2018-06-21:
my cpu is i7 6700hq and About 80 fps

PAL123, 2018-06-21:
where you have tested it?

PAL123, 2018-06-21:
[image]

PAL123, 2018-06-21:
Another good question is why the curve mapping tool does a predefined curve, even though it should / could go from point to point

this causes the above Problem

[image]

sthalik, 2018-06-21:
> Another good question is why the curve mapping tool does a predefined curve, even though it should / could go from point to point

What do you mean by "predefined curve" and "point to point"?

PAL123, 2018-06-21:
the Points where you set in the curve mapping makes often no sense, when you Combine this with other points

PAL123, 2018-06-21:
example for the weird Points:

[image]

[image]

[image]

[image]

PAL123, 2018-06-21:
the red curve should go straight as the White line, and with each additional point you can build a curve
this would definitely remove the problem and opentrack would go towards final

[image]

PAL123, 2018-06-21:
another funny curve :P 
I hope you understand what i mean. otherwise I am grateful that there are people like you who support such things

[image]

sthalik, 2018-06-22:
@PAL123,

I'm not sure how your problem touches on mappings, given I can't reproduce the problem with your exact values. But if it works for you, let's be happy and let the barking dogs lie 😄🐈.

I took the time to revisit spline widget code while investigating your issue. Now there are no jaggies on HiDPI, at least my 200% scaling. There's special debugging code that can exacerbate the issue and it no longer does.

> i mean why it goes not Point to Point straight

opentrack is using [cubic interpolation](https://en.wikipedia.org/wiki/Cubic_Hermite_spline). What you'd want is [linear interpolation](https://en.wikipedia.org/wiki/Linear_interpolation). The former is what users want.

Catmull-Rom interpolation is special in that the function goes through each point directly. Other cubic methods make it more relaxed.

> another funny curve

Unfortunately this is expected in some cases.

Also if your initial point is at `(0, 0)`, you can remove it. It's added automatically for the internal computation.

PAL123, 2018-06-22:
another example of how precise FaceTrackNoIR is in comparison to OpenTrack with almost identical settings. whether a setting is good, you can tell by the fact that it makes a natural movement from the outside back to the middle and this pretty much finds the center again 

FaceTrackNoIR
https://youtu.be/goj7YKzbTg0

OpenTrack
https://youtu.be/WDHtiNXzkFA

sthalik, 2018-06-22:
What are your current filter settings?

PAL123, 2018-06-22:
in what, Video capture filter? ?

PAL123, 2018-06-22:
FaceTrackNoIR il2b.ini
https://ufile.io/b7rvs

OpenTrack default.ini
https://ufile.io/7tl02

PAL123, 2018-06-22:
[image]

sthalik, 2018-06-22:
No, Accela settings. Also what is your LED size as displayed on the video feed?

I can't read the `.ini` binary strings same as you :)

PAL123, 2018-06-22:
but you could load it in each software

sthalik, 2018-06-22:
Can't you just alt+printscreen and ctrl+v here man?

PAL123, 2018-06-22:
yea i can, but you want it from opentrack?

sthalik, 2018-06-22:
Yes.

PAL123, 2018-06-22:
deadzone makes no difference for me, so i Keep it. but smoothing i like it on a high value, otherwise it will be much more worse

[16 more replies]

closed 2019-02-11 by sthalik