opentrack/opentrack #803: Freepie to Mouse jitter/jumps

dusan-t, 2018-06-20:
I'm using Freepie on Android to translate phone movements to mouse movements in a game, but it's never as smooth as when I'm using OpenVR-Opentrack to translate my phone movements to SteamVR. In SteamVR even without any external filter applied there is absolutely zero micro-jumps/stutters, while with regular Opentrack and mouse even with Accela the games are practically unplayable. I'm playing with a phone VR headset so even small jumps which might not be visible on a monitor are game breaking in VR. IS there anything that can be done about this ? Thanks.

sthalik, 2018-06-20:
- What's your freepie output Hz?
- Are you comparing to non-freepie?
- Does your phone have all of magnetometer, accelerometer, gyro sensors?
- Remove headphones and unplug speakers for testing.

dusan-t, 2018-06-20:
I'm using the same setup as with OpenVR driver for Opentrack (https://github.com/r57zone/OpenVR-OpenTrack/blob/master/README.md). Only difference is that when I'm using OpenVR-Opentrack my tracking data is sent from Freepie to Freetrack 2.0 and then to SteamVR - and this is where tracking is fine, while when I send the tracking data to mouse I get small jumps, like micro jitters/stutter, not sure about the exact terminology. So I don't think it's my hardware setup. Could it be that SteamVR's filters are better than Accela, or perhaps mouse isn't as precise as SteamVR's tracking ? If you still need me to post Freepie output in Hz, let me know and I will post it as right now isn't the best time for me to connect the phone to PC. Thanks.

sthalik, 2018-06-20:
I'm gonna check with the PT tracker, stay tuned.

Maybe I should implement subpixel precision for mouse movement. Right now it's limited to one-pixel movements on the screen's resolution. But I doubt that's what you're experiencing, given that the mouse tracker works fine in general.

sthalik, 2018-06-20:
A far shot–check if mouse acceleration is enabled.

sthalik, 2018-06-20:
I forgot. Try with a different game. opentrack's using raw input for sending mouse events. Maybe this game reacts badly to them.

sthalik, 2018-06-20:
- Can you try it with anything else than mouse emulation?
- Have you tried carrying out all magnets out of the room like speakers and headphones?
- Does your phone absolutely have all the gyro+mag+acc sensors? Check "debug" in freepie app and see if all three show up.

dusan-t, 2018-06-20:
That's the whole point why I've opened this issue. When I use Opentrack with OpenVR driver to send tracking data from my phone to SteamVR head tracking there is less involuntary movement than there is with the mouse. But, I've just found out that head tracking sensitivity was lower in SteamVR, so I lowered the sensitivity of the mouse to approximately level of SteamVR and mouse jumps decreased noticeably.

Btw, I'm still using Opentrack 2.3.9 because with 2.3.10 it takes much longer to establish connection over USB tether between my phone and PC while mouse in Tridef doesn't work at all. I'll try once more with the 2.3.10 before I open a separate issue.

sthalik, 2018-06-21:
> I lowered the sensitivity of the mouse to approximately level of SteamVR and mouse jumps decreased noticeably.

So how bad is it now?

If you're using tridef, why not use a non-HMD head tracking protocol in lieu of the mouse? That's gonna work way better.

By all means, please open a new issue if there's another problem.

dusan-t, 2018-06-21:
Well, it looks like I had to lower mouse sensitivity below the level of SteamVR in order to get it to acceptable level, which is a bit too slow now, but I will have to do some more testing to be sure. 

What do you mean by 'non-HMD head tracking protocol instead of the mouse' ? Games that I play accept only mouse and gamepad input, but I guess you meant different input than the phone sensors ? Can you recommend a good 3 or 6 DoF tracking solution that is better than a modern smartphone ? Thank you.

dusan-t, 2018-06-21:
Or did you mean to say I should look into abandoning head tracking completely ? I don't want that, I'd like to keep the option of looking around with my head and have that translated to in-game camera movement.

sthalik, 2018-06-21:
> Games that I play accept only mouse and gamepad input

Use vjoystick output if you can set head tracking to gamepad.

dusan-t, 2018-06-22:
Tried setting head tracking to gamepad and it's even worse than mouse, but in a different way. It's lags so much that it's ridiculous, even with controller sensitivity set to max. But I managed to get mouse to an acceptable level in 2 games out of 3 that I was playing. It seems it depends on the game too.

sthalik, 2018-06-22:
Does it look any sensibly on desktop?

dusan-t, 2018-06-22:
Actually desktop is the only way I tried it, since how laggy and inappropriate for the purpose it was, there was no point in trying it with the headset. Regarding the mouse imprecision though, I'm wandering if maybe that's how it is in general and not just in my case. I watched a video from a guy who made StreamTheater app for GearVR and he also mentioned lowering sensitivity to the lowest to avoid mouse imprecision. As for output to SteamVR being more precise than output to mouse, well SteamVR was built with head tracking in mind, mouse was not. 

There's an official document written by Oculus on how they solved head tracking problems: msl.cs.illinois.edu/~lavalle/papers/LavYerKatAnt14.pdf

They go into detail, I think it's worth a read.

sthalik, 2018-06-22:
No, try looking at the cursor on the Windows desktop. See if it jumps around. If you keep the window focused, space bar will press the "stop tracking" button.

dusan-t, 2018-06-22:
Oh, I thought you were still referring to vjoy/gamepad approach. Since I'm only playing in Tridef with my headset I don't have desktop in SBS so I can't check how jumpy it is on the desktop while having the headset on my head, I can only speak of how it looks when I hold it in my hand which isn't accurate representation as my hands seem to move/tremble, at least more than my head does. To test this I would have to install TrinusVR and have my desktop show up inside my headset/phone in SBS. I will post if/when I do that.

sthalik, 2018-06-22:
Well you can either

- mount on top of your head
- mount on top of your head with duct tape

closed 2018-08-03 by sthalik