opentrack/opentrack #804: Prepar3d V4 issues with simconnect

MathijsG, 2018-06-21:
Hi Sthalik,

I'm trying to get Opentrack working with Prepar3D V4.

With the latest patches and with the latest V4 SDK version: 4.2.21.24048

However none of the entries of the FSX Simconnect output protocol work. I even tried installing all the different version at the redist folder of Prepar3D (not all at once ofcourse), but I still get this very familiar error:

[image]

FaceTrackNoIR does work, it has a Simconnect V2 protocol build in, since a few releases back, however I get low framerates with that somehow.

Do you have any direction for me to dive in or something which can help?

Where freetrack (meaning the protocol) always works without a hassle in every game, the simconnect protocol sounds like a little hell :-P.

sthalik, 2018-06-21:
@MathijsG,

There's a workaround to get simconnect working in the last stable version. Copy the simconnect `.dll` into one with the same name but no `lib` prefix.

But better use the unstable version. It's almost always less buggy than the stable version.

sthalik, 2018-06-21:
Been fixed in unstable. Closing.

MathijsG, 2018-06-21:
Do you have the compiled unstabled versions somewhere? Can't seem to find em on /releases/

sthalik, 2018-06-21:
http://opentrack.misaki.pl

MathijsG, 2018-06-21:
Thanks, that did the trick!

sthalik, 2018-06-22:
I'll be releasing a new version soon. Can you help with last-minute tests?

MathijsG, 2018-06-22:
Yeah, guide me master.

MathijsG, 2018-06-26:
@sthalik specific things where I can help?

sthalik, 2018-06-26:
@MathijsG I'm releasing a new build as soon as I commit individual changes. Can you post in #778?

The changes are largely internal (not user-visible) but there are some good nuggets here and there.

There's some new fancy stuff in the first post's changelog too.

sthalik, 2018-06-27:
@MathijsG see https://github.com/opentrack/opentrack/issues/778#issuecomment-400865602

sthalik, 2018-06-27:
I really should test this PT feature more. @MathijsG  Can you see if it helps reduce the amount of noise, or is it worse (more filtering needed)?

JDTiger, 2019-01-19:
Hey, i donwloaded the latest build from the your website, but I still get the simConnect error trying to use it with P3Dv4

MathijsG, 2019-01-20:
@JDTiger Did you use this website: http://opentrack.misaki.pl/ and downloaded the latest?

Did you try every simconnect version?

sthalik, 2019-01-20:
@JDTiger install the simconnect sdk that comes with fsx/p3d.

JDTiger, 2019-01-20:
Simconnect should be installed, since it works with other plugins...
Yes i did download the version from the website and i treid every simconnect version

sthalik, 2019-01-20:
What does opentrack version say on its title bar?

JDTiger, 2019-01-25:
Are there any updates on my problem?
Or anything i can provide you/ try out?

sthalik, 2019-01-26:
I don't have a build of P3D at the moment. What is your P3D version? Was there any new update?

Have you run anything "SimConnect" related and had it succeed? In fact the component you need to install is called "SimConnect SDK". The name is misleading as if it was meant for developers, but every FSX/P3D user needs it for plugins to work.

Let me grab some P3D and do a basic test.

sthalik, 2019-01-28:
From <<https://www.avsim.com/forums/topic/518492-simconnect-in-p3d-v4/>>,
> You will find all versions in Prepar3D v4\redist\Interface folder. Just run every installer there, except FSX-RTM-JPN and FSX-SP1-RUS versions. They need to be installed manually for some other addons, so i think this can be your problem.

JDTiger, 2019-01-28:
Ok, I installed all versions of SimConnect and created the .cfg and .xml files, still get the errors...

JDTiger, 2019-02-03:
Ok, i do not get the errors anymore, though the head ingame does not move...
Or am i just to blind to finde the ingame enable button? :D

Jabby8, 2019-02-06:
Hi
same issue for me, with P3Dv3 (3.4.18.19475)
There are no more error message with latest build (opentrack-test-20190203_01) and tracking is on, but no move in game.
;)

sthalik, 2019-02-09:
Have you enabled 3D cockpit? It's been ages since I used anything FSX-related but it's that way in X-Plane.

JDTiger, 2019-02-09:
I tried running both apps as admin, played around with the view settings and tried changing the bindings, but it was not possible to change the bindings by moving the head.

sthalik, 2019-02-10:
Hey,

It was a bug. It should now work with P3D.

sthalik, 2019-02-10:
See today's build.

JDTiger, 2019-02-10:
Yes, it works now, thank you

Jabby8, 2019-02-10:
Perfect, it also works on P3Dv3
Thank you

closed 2019-02-11 by sthalik