opentrack/opentrack #805: Check permissions for your .ini directory

InsanelyOrange, 2018-06-24:
After installing and launching open track for the first time, I was greeted with this error message.
[image]
As you can see, the folder exists and I (along with all other groups have full access to the file). 
Running Windows 10, not sure what's going on.

sthalik, 2018-06-24:
See if you checked "Windows Defender -> Controlled folder access" (aka ransomware protection). It's off by default.

Did you create the directory by yourself or was it created on its own? The latter should happen. But in the same case, the missing `.ini` file should get created as well!

This logic can happen either before the main window shows up, or during program runtime. I retraced it and it's sound.  You're also the first person to report this.

InsanelyOrange, 2018-06-24:
> See if you checked "Windows Defender -> Controlled folder access" (aka ransomware protection). It's off by default.

I'm running BitDefender as my antivirus and turned off ransomware protection but the error still persists. I tried uninstalling and reinstalling with ransomware protection off and no dice.

> Did you create the directory by yourself or was it created on its own? The latter should happen. But in the same case, the missing .ini file should get created as well!

The folder was created on it's own, strange the ini wasn't created.

> This logic can happen either before the main window shows up, or during program runtime. I retraced it and it's sound. You're also the first person to report this.

Never been able to get the program to open, so I guess I'm experiencing the former part of the logic.
I'm the lucky first, Saw another post about this on some Spanish or Portuguese flight sim board, that was it, haha. Thanks for taking the time to respond.

sthalik, 2018-06-24:
But I'm not done! I intend to help you with the problem.

Antivirus software normally has an option to disable it for X minutes in the tray icon. Can you try with that first?

InsanelyOrange, 2018-06-24:
> But I'm not done! I intend to help you with the problem.

Glad to hear it since you've helped me solve my problem. It seems Bitdefender was blocking opentracker from writing files to my Documents folder. I simply went to View Features -> Safe Files -> Application Access and added opentrack.exe as an exception. opentrack now launches successfully. 

Thanks a lot for your help sthalik, I probably should've looked at my anti virus software before bothering you!

sthalik, 2018-06-24:
I'm glad this helped! Cheers.

closed 2018-06-24 by sthalik