opentrack/opentrack #807: Just another noob question on FreeTrack and x64 games

HornetMaX, 2018-07-03:
Hi all,
I havew been using OpenTrack with an EdTracker in some bike sims (GP Bikes, MX Bikes) with excellent results via FreeTrack interface. Recently MX Bikes went x64 and now I'm stuck.

What are the options here ? get a freetrack64.dll ? Maybe using TrackIR interface (as it seems OpenTrack does provide the x64 dll for that) ?

sthalik, 2018-07-03:
I don't believe there's any 64-bit "freetrackclient" support. Even if opentrack provided that, nothing uses it.

NPClient64.dll is known to work across many titles. That use it, of course.

HornetMaX, 2018-07-04:
Well I guess apps used to freetrackclient.dll going to x64 would gladly have a freetrackclient64.dll instead of switching to something else :)
Anyway, I get that your suggestion is for the hosting x64 app to simply use NPClient64.dll instead. Thx !

jp7677, 2019-10-21:
> I don't believe there's any 64-bit "freetrackclient" support. Even if opentrack provided that, nothing uses it.
> 
> NPClient64.dll is known to work across many titles. That use it, of course.

Hi @sthalik 
I'm sorry for resurrecting this old issue. Unfortunately Elite Dangerous/Horizons x64 (via SteamPlay/Proton/wine on Linux) seems to want a freetrackclient64.dll, at least on my system . Looking at the wine logs the game loads `NPClient64.dll` but also tries to load `freetrackclient64.dll` which it cannot find. Head-tracking didn't worked on my system, but started to work immediately once I placed `freetrackclient64.dll` in the `opentrack/libexec` folder. I took the `freetrackclient64.dll` from the setup package provided at https://sourceforge.net/projects/facetracknoir/files/
Could you please consider including a `freetrackclient64.dll` like you did with the `freetrackclient.dll`?

sthalik, 2019-10-21:
@jp7677 I'll look into it.

commit 29591277 (2019-10-30): options: use ~/.config on Unix

Fixes #807
closed 2019-10-30 by sthalik