opentrack/opentrack #808: Green only in Color channels used?

DanTDBV, 2018-07-08:
First of all.

Thank you very much for this great piece of software!

As you have asked people with blue LED’s to pipe up, I have decided to do this as well.;)

I am using green LED’s for my track cap
I would therefor be grateful if a Green only choice, could be added to Color channels used in the PointTracker settings section of PointTracker 1.1.

My setup is triple screens together with a Logitech C920 webcam and a DIY track cap with 3 green LED’s, using PointTracker 1.1, freetrack 2.0 Enhanced and Accela. 

To gain a better understanding of opentrack, I am currently working on a mostly personal guide, for the parts that I am going to use of the software. When I finish I intend to share it with you and the community.

If you find any of it useful, you will be very welcome to use it.

Where would the best place to show what I have gotten so far, so as to confirm if I have understood it correctly. I would also like to know, the best place to ask any questions, when I get stuck.

I will wait with questions until I have written up on the stuff, that I think I understand.

Best regards from Denmark

DanTDBV (The Droid Building Viking)

DanTDBV, 2018-07-08:
Having read 796 about donation, I will say that you should only do this if it is fast and easy to do. I have it working without, so your time is better used on the more important stuff. 

If others also wish it, it is another story.

I am absolutely not a coder guy, so unfortunately I cannot help you there.

I will however continue with the guide. Now I have also found a great guide over at the IL-2 Sturmovik forum and I have already found some of my answers there. I will ask my other questions there first.

The Sturmovik guide
https://forum.il2sturmovik.com/topic/34403-a-complete-guide-to-set-up-head-tracking-opentrack/

Have a nice day.

sthalik, 2018-08-03:
Get today's build `20180803_02` from <<http://opentrack.misaki.pl>>. It has to do between releases.

closed 2018-08-03 by sthalik