opentrack/opentrack #809: Opentrack axis issue

BlakerDCS, 2018-07-08:
Hello. I am new on opentrack because I use facetracknoir, but I have a problem. When iI want to go in direction of my camera (Z axis), My X axis is going right. Opentrack is detecting my trackclip going right for approx 2 degrees. It's really annoying because when I want to be centered, I have to go left. Facetracknoir doesn't make that. I've try and my X axis is not moving on Facetracknoir.

Please help, I am on it for 5 hours ! I really want to throw this software through the window XD.

sthalik, 2018-07-08:
It's impossible to be perfectly centered. I'm adding a calibration function for the Z axis already. It just hasn't been done yet.

For now use less sensitive mappings for the Y axis and some others.

BlakerDCS, 2018-07-09:
ok thanks for the answer, I'll keep facetracknoir until the update.

closed 2018-08-04 by sthalik