opentrack/opentrack #811: relative translation

albertolg, 2018-07-10:
@sthalik I haven't used opentrack for a while, (no time to fly), downloaded latest test build and I noticed that the relative translation is working different from the release build 2.3.10 of feb 28. What is changed? I can't use latest test build because the beaviour is very unrealistic... will you restore the relative translation as was before?

albertolg, 2018-07-10:
well, same settings in both versions... reltrans enabled... 
I hope I can explain the difference.... with latest versions if I look at left or right and move my head in that direction, like I want to look outside the plane's cockpit, then the view doesn't move in that direction I'm looking at but will move forward or backward in the cockpit (the Z axis and not Y as should do, so will slide left or right)... If is not clear my explanation I'll make a video, but with the 2.3.10 it works fine as should do...

sthalik, 2018-07-11:
I understand, and will check all axis.

sthalik, 2018-11-09:
@albertolg, if you wanted to move outside the window with reltrans, you should, for left:

- rotate left to 45-120 deg as needed
- move toward the screen

Is this what you're doing?

albertolg, 2018-11-09:
> 
> 
> @albertolg, if you wanted to move outside the window with reltrans, you should, for left:
> 
>     * rotate left to 45-120 deg as needed
> 
>     * move toward the screen
> 
> 
> Is this what you're doing?

yes, exactly, and with the 2.3.10 is working, but with later builds, not

sthalik, 2018-11-11:
Unfortunately it works for me.

I rotated by some 45 deg left, then leaned toward the screen. The X value changed with minor changes to Z. Same happened when rotating right and leaning forward.

Verify whether your leaning forward is recognized as such by:

- disabling relative translation temporarily
- leaning forward and observing XYZ values

It's the Z value that should change. It's possible that X and Y values change more.

sthalik, 2018-11-11:
Please use the latest snapshot from http://opentrack.misaki.pl for that test.

albertolg, 2018-11-11:
ok, did a new profile from scratch, now is working well, probably I had problems because I imported an old one?

closed 2018-11-11 by albertolg