opentrack/opentrack #812: Yaw and magnetometer raw data

Su33ke, 2018-07-11:
Hi! I make an application that calculates the yaw pitch roll from the accelerometer and magnetometer and sends it to opentrack by udp. Everything works well if I play face to the north. But if I play facing east and centering the camera, the raw yaw will be -90. As a result, the range to the right will be 90 degrees instead of 180, because after -180 there will be 180. What can I do about it? Sorry for my terrible English.

Su33ke, 2018-07-11:
Sorry, this is a very stupid issue :)  I just added yaw = yaw + 90 in my app.
What do you think about adding the possibility of raw data adjustment in opentrack? Because the way that I found is not the best solution, I think

sthalik, 2018-07-12:
I assumed it works after switching signs. What happens then?

Su33ke, 2018-07-12:
After changing the sign, he decides that I turned left. Maybe I'm doing something wrong. Now I will record a video.. 
[https://www.youtube.com/watch?v=IOmUzzwD_dw&feature=youtu.be](url)

Thank you for your wonderful project and support!

sthalik, 2019-01-23:
It should work now. Can you confirm this please?

Su33ke, 2019-01-23:
Sorry, I don’t have a suitable device to check now.