opentrack/opentrack #813: single axis translation affects others

dekustronk, 2018-07-11:
Hello.
After 'successfully' calibrating the model position I stumbled upon a problem when **translating side-to-side** on the X axis. Moving on this axis would also introduce **changes** on the Y axis and a bit of Yaw rotation.

The problem **might** have to do with my 3-point clip's default position, which is slanted 20 degrees from vertical. I tried using a custom cap model and changing the points positions by rotating their relative coordinates by 20 degrees, but the result is that it would not calibrate near the head pivot at all and the movement was erratic even when manually adjusted.

Here is a link to a ~20 sec gif displaying the behaviour (side to side at around 10 sec): https://gfycat.com/gifs/detail/SlightComposedArcherfish
The rotations appear to work fine, as well as moving on the Y and Z axis, the more apparent problem for me seems to be the X axis.

Regarding the custom model with rotated points for the standard clip, if I calculated correctly, they should be (-28, 28, 30) and (49, -50, 80) and the middle one is the origin.

Len62, 2018-07-12:
Have you tried adding a small dead zone for the Y axis and yaw in the mapping section? Any small Y movement, be it translation or rotation, while moving on the X translation gets picked-up by opentrack.

dekustronk, 2018-07-12:
But when moving side to side, the yellow cross remains almost at the same level even in raw tracker data. The problem is that even after it calibrated fine it still introduces a lot of movement on the Y axis.

Len62, 2018-07-12:
The yellow cross does not produce any raw tracker data, only the tracked LEDs. Without a little dead space even the **slightest** movement of your head can have an effect on other axis movement. Please try adding some dead space and see what happens. Also, you can adjust sensitivity in the Accela settings.

dekustronk, 2018-07-12:
I set up some deadzone for the Y axis but then it becomes unusable. As I said before, if you look at that gif you will see that the movement barely affects the Y raw tracker data with 1 unit. I tried to bend my head and try to get the clip aligned vertically and then move and it seems to work better when moving on the X axis.

I believe that the 20 degree default position rotation of my clip has to do with it. Does opentrack account for such rotations? I tried rotating all the coordinates in my clip by using the custom model but it didn't work out that well.

sthalik, 2018-07-12:
I don't recommend using translation increasing that fast. Also, rather than using a deadzone on the Y mapping, try adding a few points so the mapping increases faster after a while.

dekustronk, 2018-07-12:
I lowered the sensitivity of those axis and also I am going to mainly use the angling capabilities. I searched here for older issues similar to this and I couldn't find anything conclusive to solve this.

I still think that it has to do with the default angle of 20 degrees (the error introduced on the Y axis is symmetrically going in opposite directions when moving left vs right). Would the custom model tab work if I tried tinkering with it more? Will you introduce some kind of error correction for this in the future?

Thanks.

sthalik, 2018-07-12:
It's that the calibrator can't arrive at a solution that always works perfectly no matter the circumstances.

dekustronk, 2018-07-12:
I tilted my head so that the clip is straight. Now the movements seem to work better so this is why I think the unwanted movements on the other axis are more than an error in auto calibration, but have something to do with the tilt.

gif: https://gfycat.com/DelayedPaleDamselfly (HD button if not on by default)

sthalik, 2018-07-12:
Can you compare to 2.3.9?

dekustronk, 2018-07-12:
Do you have a link to an installer?
I just recently started using opentrack and I don't know how to navigate github that well (made an account to comment here).

dekustronk, 2018-07-12:
Same problem when moving side to side with clip tilted (normal position) in 2.3.9 .

If I were to use a custom model, other than rotating the coordinates of the points and head position with a rotation matrix 20 degreees counter clockwise, do I need to do something more?

dekustronk, 2018-07-13:
@sthalik I corrected the values in the custom model to make the clip appear 'tilted' by default and it now works correctly. 
I suggest that in the future you should implement a feature into the already existing center command to **actually capture the clip in its default position**. 
The errors that the raw data show now are minimal so I won't compensate for pitch and yaw, but the program should take such deviations into account as they do not seem to be part of the auto calibration margin of error but have to do with not accounting for the skewed variation from what the program considers as the standard 3-point clip (be it roll, pitch or yaw).

sthalik, 2018-08-04:
> I suggest that in the future you should implement a feature into the already existing center command to actually capture the clip in its default position.

The program takes a transpose of rotation matrix when center is ordered.

I agree about the skew. See #704.

closed 2018-08-04 by sthalik