opentrack/opentrack #814: Raw tracker data and game data don't match up

quinn1005, 2018-07-12:
I havent really screwed with anything to do with changing the output but for some reason the raw and game data are no where near each other. I am using a aruco tracker on my headset, it was working just fine until i tried to put my phone in via trinus. I tried turning off trinus but it's still doing it 
[image]

sthalik, 2018-07-12:
Your marker is rotated. First make sure raw data roll is close to zero.

quinn1005, 2018-07-12:
ok i'll try that

quinn1005, 2018-07-12:
ok it works now thanks man. Just need to get a better webcam now, my kinda blurs if i move fast so it freezes then jerks to where im looking when it can recognize the qr code again lol

sthalik, 2018-07-12:
Just so you know, PS3 Eye works worse with Aruco worse than a 30 Hz camera.

closed 2018-07-12 by sthalik